tbnobody/OpenDTU #1688: power of the line remains at its last value after stopping the power supply to the line

Ricko25000, 2024-01-26:
### What happened?

at night I use an HM-700 on battery with current regulation on line 2 (45Vdc 2.7A = 120W). a relay cuts the power supply to line 2 as well as the 230AC line.
in the morning, after the cutoff, line 2 remains at its last power of 120w and opendtu continues to accumulate this power while the HM-700 is no longer powered.

### To Reproduce Bug

have power on line 1 or 2 and cut the AC line suddenly.
that the power on line 1 or 2 is at 0w when the inverter is no longer communicating.

### Expected Behavior

when there is no more power on line 1 and 2 set the power to 0W

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	v24.1.21

tbnobody, 2024-01-26:
Have you enabled this parameter?
[image]

Ricko25000, 2024-01-27:
it's ok with your setting.
THANKS. you are doing a great job.

closed 2024-01-27 by Ricko25000