tbnobody/OpenDTU #1694: Display SSD 1306 - White line on left side

HanSolu123, 2024-01-28:
### What happened?

Hello,

there is a white line on the left side on the display (with rotation of 180°.
Anybody knows how to eliminate ?
[image]

### To Reproduce Bug

Always
SW Version 24.1.26 and earlier

### Expected Behavior

Elimante the white line

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

24.1.26

tbnobody, 2024-01-28:
Seems like the wrong display type is selected. SH1106 vs SSD1309

HanSolu123, 2024-02-03:
Thanks for your tip. Changing to the correct display type solved the probelm

closed 2024-02-03 by HanSolu123