tbnobody/OpenDTU #1707: DTU-Einstellungen Frequenz funktioniert nicht.

FrodoVDR, 2024-01-31:
### What happened?

Ich habe die Frequenz 867 MHz eingestellt und bekomme die Warnung das dies außerhalb des in der EU zulässigen Bereichs liegt. Hierbei ist der Schieber ganz rechts. 
Verändere ich den Schieber ist ganz Rechts eine Frequenz von über 900 MHz.
Gehe ich nun auf meine 867 MHz, werde ich nach dem speichern gewarnt, nun das die Frequenz in 250 kHz Schritte erfolgen muss.

### To Reproduce Bug

[image]
[image]

### Expected Behavior

Die Warnungen sollten nur erfolgen wenn die Einstellungen falsch sind.
Der Schieberegler sollte die eingestellte Position anzeigen und nur dann Fehlermeldungen ausgeben wenn die eingestellten Werte falsch sind.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.1.26

tbnobody, 2024-01-31:
Duplicate of #1651 and #1662
Just reload the WebUI

FrodoVDR, 2024-02-01:
Thanks, this works for me