tbnobody/OpenDTU #1711: Firmware upgrade stuck in upgrade process

home-cloud, 2024-02-01:
### What happened?

I try to upgrade the firmware. 
From version v24.1.21 to v24.1.26
The upgrade process stucks every time between 4% and 72%.
A reset before the update is not helping.  

### To Reproduce Bug

every try to update 

### Expected Behavior

100% to proceed

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

a5cc0a4

tbnobody, 2024-02-01:
Did you try a power cycle?

home-cloud, 2024-02-01:
> Did you try a power cycle?
yes

Josy68, 2024-02-06:
Same problem. How can I fix it? Thanks

home-cloud, 2024-02-06:
Powercycle, Login to the fallback accesspoint, Login and try to upgrade. Now i am on the latest version.

Josy68, 2024-02-06:
In access point mode the same. Go to 20%  and than the unit reboot. The old version is stored

puni2k, 2024-02-15:
My ESP had the same issue for a long time last year. Updates often got stuck at various percentages and were only successful after multiple tries.
After improving my ESPs power supply during my investigation of #1339 (more capacitors and removing a diode between the USB port and voltage regulator) that problem never showed up again.

tbnobody, 2024-10-15:
Please doublecheck if the issue still exists in the latest version

tbnobody, 2025-02-07:
Going to close this issue due to no activity and a lot of code changes since then. Feel free to open a new issue if the problem persists.

closed 2025-02-07 by tbnobody