tbnobody/OpenDTU #1725: Guru Meditation (StoreProhibited) when saving DTU settings

schlimmchen, 2024-02-06:
### What happened?

I found a Guru Meditation on a newly installed device (flash erased, only WiFi and device profile configured, only an NRF24 odule) when trying to save the DTU settings:

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401a785a  PS      : 0x00060e30  A0      : 0x80103626  A1      : 0x3ffb21a0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00001999  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x3f436bb4  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x68108001  A13     : 0x00000015  
A14     : 0x00000001  A15     : 0x3ffb0320  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x40084a45  LEND    : 0x40084a4d  LCOUNT  : 0x00000027  

Backtrace: 0x401a7857:0x3ffb21a0 0x40103623:0x3ffb21c0 0x400e3a58:0x3ffb21f0 0x401beff9:0x3ffb2210 0x400f210a:0x3ffb2230 0x400f2285:0x3ffb2250 0x400f229e:0x3ffb2270 0x4010d999:0x3ffb2290
```

This is the backtrace:
```
0x401a7857: CMT2300A::setFrequencyBand(FrequencyBand_t) at /home/schlimmchen/OpenDTU/lib/CMT2300a/cmt2300wrapper.cpp:257
0x40103623: HoymilesRadio_CMT::setCountryMode(CountryModeId_t) at /home/schlimmchen/OpenDTU/lib/Hoymiles/src/HoymilesRadio_CMT.cpp:242
0x400e3a58: WebApiDtuClass::applyDataTaskCb() at /home/schlimmchen/OpenDTU/src/WebApi_dtu.cpp:37
0x401beff9: void std::__invoke_impl<void, void (WebApiDtuClass::*&)(), WebApiDtuClass*&>(std::__invoke_memfun_deref, void (WebApiDtuClass::*&)(), WebApiDtuClass*&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/invoke.h:73
 (inlined by) std::__invoke_result<void (WebApiDtuClass::*&)(), WebApiDtuClass*&>::type std::__invoke<void (WebApiDtuClass::*&)(), WebApiDtuClass*&>(void (WebApiDtuClass::*&)(), WebApiDtuClass*&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/invoke.h:95
 (inlined by) void std::_Bind<void (WebApiDtuClass::*(WebApiDtuClass*))()>::__call<void, , 0u>(std::tuple<>&&, std::_Index_tuple<0u>) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/functional:400
 (inlined by) void std::_Bind<void (WebApiDtuClass::*(WebApiDtuClass*))()>::operator()<, void>() at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/functional:484
 (inlined by) std::_Function_handler<void (), std::_Bind<void (WebApiDtuClass::*(WebApiDtuClass*))()> >::_M_invoke(std::_Any_data const&) at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
0x400f210a: std::function<void ()>::operator()() const at /home/schlimmchen/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
0x400f2285: Scheduler::execute() at /home/schlimmchen/OpenDTU/.pio/libdeps/generic_esp32/TaskScheduler/src/TaskScheduler.h:1544
 (inlined by) Scheduler::execute() at /home/schlimmchen/OpenDTU/.pio/libdeps/generic_esp32/TaskScheduler/src/TaskScheduler.h:1344
0x400f229e: loop() at /home/schlimmchen/OpenDTU/src/main.cpp:157
0x4010d999: loopTask(void*) at /home/schlimmchen/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

### To Reproduce Bug

I guess the only requirement is a device with only an NRF24 module. Try to save the DTU settings through the WebUI.

### Expected Behavior

No Guru Meditation.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

21ec72f4c00fab1385b1fb93780b52cd6d4e4ff7

tbnobody, 2024-02-06:
Thanks for the hint. Will be fixed in the next release. If you like to test it (already tested here) you can try this patch

```diff
diff --git a/lib/Hoymiles/src/HoymilesRadio_CMT.cpp b/lib/Hoymiles/src/HoymilesRadio_CMT.cpp
index bd831eeb..bbd31f21 100644
--- a/lib/Hoymiles/src/HoymilesRadio_CMT.cpp
+++ b/lib/Hoymiles/src/HoymilesRadio_CMT.cpp
@@ -239,8 +239,11 @@ CountryModeId_t HoymilesRadio_CMT::getCountryMode() const

 void HoymilesRadio_CMT::setCountryMode(const CountryModeId_t mode)
 {
-    _radio->setFrequencyBand(countryDefinition.at(mode).Band);
     _countryMode = mode;
+    if (!_isInitialized) {
+        return;
+    }
+    _radio->setFrequencyBand(countryDefinition.at(mode).Band);
 }

 uint32_t HoymilesRadio_CMT::getInvBootFrequency() const
```

schlimmchen, 2024-02-11:
Thanks for sharing your fix. I have not tested it on the actual hardware, but I can clearly see how that fixes the issue. Once you drop your next release I will test and close this issue, if you don't do so before me.

commit 3cf8fea8 (2024-02-12): Fix: Guru Meditation (StoreProhibited) when saving DTU settings

Fixed #1725