tbnobody/OpenDTU #1747: Python script for zero export does not run anymore with 24.2.12

MarkusPVHM, 2024-02-14:
### What happened?

Runing the python script with the firmware 24.2.12, there is a error no data from DTU. Is there changed somehing in the json export?

### To Reproduce Bug

every time, fallback to previous version everthing works

### Expected Behavior

make the script from 

https://github.com/Selbstbau-PV/Selbstbau-PV-Hoymiles-nulleinspeisung-mit-OpenDTU-und-Shelly3EM
run again swith new firmware

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.2.12

MetaChuh, 2024-02-14:
@MarkusPVHM 

the api has changed.

a multi pass query is now necessary.
this has been implemented for memory consumption and stability reasons, especially if using a larger number of inverters, as the esp32 memory is small.

please use opendtu v24.1.26 until zeroexport has been updated.

basic info:
at the endpoint `/api/livedata/status` you get the inverter list and basic information.
then to get the inverter data, add the serial to the next request `/api/livedata/status?inv=116480148266`

see:
[Breaking API Changes](https://github.com/tbnobody/OpenDTU/discussions/1463)

example:
https://github.com/tbnobody/OpenDTU/issues/1615#issuecomment-1902468777

ps: @reserve85 : api changes everywhere 😉

reserve85, 2024-02-14:
Hi, I updated it already. The new version is in the dev path, please use this. https://github.com/reserve85/HoymilesZeroExport/tree/dev

@MetaChuh @MarkusPVHM

tbnobody, 2024-02-16:
In this case I would close the issue.

closed 2024-02-16 by tbnobody

MarkusPVHM, 2024-02-16:
> In this case I would close the issue.

sorry, I am using another scipt von Selbstbau-PV.  Therefore my problem isn't solved.

https://github.com/Selbstbau-PV/Selbstbau-PV-Hoymiles-nulleinspeisung-mit-OpenDTU-und-Shelly3EM

This one is it....

reserve85, 2024-02-16:
Then you should open an issue in their repository. It’s not a problem of OpenDTU.