tbnobody/OpenDTU #1751: Support for HERF-800 inverter (kind of already working)

Birkenstab, 2024-02-15:
### Describe the solution you'd like

The HERF-800 inverter from e-star is using the same wireless communication as the Hoymiles HM-800-2T.
Unfortunately we cannot enter the serial number of the Herf inverter because OpenDTU does not know which device type it should use and we cannot enter hex digits or alphanumeric digits (only 0-9 are possible) in the web gui. Therefore we need to transform and modify our serial number.

Lets assume the serial number of the inverter is `A1100FBT4-19W`. We can ignore the 'extend code' `-19W` at the end.
Thankfully rovo89 wrote a script on [Discord](https://discord.com/channels/984173303147155506/1146388875204378636/1168635100527870023) to convert this serial number from 9 digits alphanumeric to 12 digits hex:
```python
sn = 'A1100FBT4'
sn_int = 0
CHARS = '0123456789ABCDEFGHJKLMNPRSTUVWXY'
for i in range(0, 9):
  pos = CHARS.find(sn[i])
  shift = 42 - 5 * i - (0 if i <= 2 else 2)
  sn_int |= pos << shift
print(format(sn_int, 'X'))
```

For our example serial number this outputs: `28210007AF44`
Now we need to replace the first digits by 1141: `11410007AF44`
Why 1141? Because HM-800 inverter serial numbers start with 0x1141: https://github.com/tbnobody/OpenDTU/blob/master/lib/Hoymiles/src/inverters/HM_2CH.cpp#L44

As of right now the web gui of OpenDTU does not allow us to enter hex digits, so we just create an inverter with an arbitrary serial number. Does not matter. 

Now we need to convert our hex serial number `11410007AF44` to decimal: `18970871050052`.
Then we download a backup of the config.json in `configuration management` and edit the json file.
Under `inverters` we search for our newly created inverter and replace the `serial` with our converted decimal serial number :
```json
…
 {
            "serial": 18970871050052,
            "name": "test",
            "order": 0,
            "poll_enable": true,
            "poll_enable_night": true,
            "command_enable": true,
            "command_enable_night": true,
            "reachable_threshold": 2,
…
}
…
```

Now we can upload the config.json to OpenDTU by restoring the configuration.
Under settings > inverters the inverter should now be listed as type `HM-600/700/800-2T` with our modified serial number as hex:
[image]

And it should also work as expected (with a few exceptions, see at the end):
[image]

[image]

Even the eventlog is working.

Apparently the preId of the serial number (0x1141) is not used to identify the inverter.

## Whats not working
* Creating the inverter via web gui. Hence this guide
* Setting power limit, restart and enable/disable inverter in the web gui. Apparently the frontend (client and server side) does not like serial numbers that include a-f digits). Doing it via Home Assistant/MQTT seems to be working fine
* Sometimes the power limit is not displayed correctly

Maybe I can work on fixing those things and generally making it possible to create the inverter in the gui in the next few days

tbnobody, 2024-02-15:
Could you  apply this patch  to see if power limit etc. now works as expected?

```diff
diff --git a/src/WebApi_dtu.cpp b/src/WebApi_dtu.cpp
index aed09834..724f922f 100644
--- a/src/WebApi_dtu.cpp
+++ b/src/WebApi_dtu.cpp
@@ -129,7 +129,7 @@ void WebApiDtuClass::onDtuAdminPost(AsyncWebServerRequest* request)
         return;
     }
 
-    if (root["serial"].as<uint64_t>() == 0) {
+    if (strtoll(root["serial"].as<String>().c_str(), NULL, 16) == 0) {
         retMsg["message"] = "Serial cannot be zero!";
         retMsg["code"] = WebApiError::DtuSerialZero;
         response->setLength();
diff --git a/src/WebApi_inverter.cpp b/src/WebApi_inverter.cpp
index 32a47235..d837c10b 100644
--- a/src/WebApi_inverter.cpp
+++ b/src/WebApi_inverter.cpp
@@ -129,7 +129,7 @@ void WebApiInverterClass::onInverterAdd(AsyncWebServerRequest* request)
         return;
     }
 
-    if (root["serial"].as<uint64_t>() == 0) {
+    if (strtoll(root["serial"].as<String>().c_str(), NULL, 16) == 0) {
         retMsg["message"] = "Serial must be a number > 0!";
         retMsg["code"] = WebApiError::InverterSerialZero;
         response->setLength();
@@ -233,7 +233,7 @@ void WebApiInverterClass::onInverterEdit(AsyncWebServerRequest* request)
         return;
     }
 
-    if (root["serial"].as<uint64_t>() == 0) {
+    if (strtoll(root["serial"].as<String>().c_str(), NULL, 16) == 0) {
         retMsg["message"] = "Serial must be a number > 0!";
         retMsg["code"] = WebApiError::InverterSerialZero;
         response->setLength();
diff --git a/src/WebApi_limit.cpp b/src/WebApi_limit.cpp
index 1d9c111a..d196c895 100644
--- a/src/WebApi_limit.cpp
+++ b/src/WebApi_limit.cpp
@@ -100,7 +100,7 @@ void WebApiLimitClass::onLimitPost(AsyncWebServerRequest* request)
         return;
     }
 
-    if (root["serial"].as<uint64_t>() == 0) {
+    if (strtoll(root["serial"].as<String>().c_str(), NULL, 16) == 0) {
         retMsg["message"] = "Serial must be a number > 0!";
         retMsg["code"] = WebApiError::LimitSerialZero;
         response->setLength();
diff --git a/src/WebApi_power.cpp b/src/WebApi_power.cpp
index b5196789..6675da6c 100644
--- a/src/WebApi_power.cpp
+++ b/src/WebApi_power.cpp
@@ -93,7 +93,7 @@ void WebApiPowerClass::onPowerPost(AsyncWebServerRequest* request)
         return;
     }
 
-    if (root["serial"].as<uint64_t>() == 0) {
+    if (strtoll(root["serial"].as<String>().c_str(), NULL, 16) == 0) {
         retMsg["message"] = "Serial must be a number > 0!";
         retMsg["code"] = WebApiError::PowerSerialZero;
         response->setLength();

```

tbnobody, 2024-02-15:
> rovo89: Übrigens sind die ersten vier Stellen wohl der Gerätetyp (A=MI, B=ES, C=RSD, D=DTU), dann zwei Stellen für den OEM, dann eine Stelle für die Fabriknummer. Die restlichen 5 Stellen sind eine Sequenznummer. Plus dann halt ein Bindestrich und der "Extend Code".

This should be wrong as your inverter seems to be a HM instead of MI

tbnobody, 2024-02-15:
And btw... this snipped from the linked issue and your serial number gives your `-19W`
```python
sn = 'A1100FBT4'
CHARS32 = '0123456789ABCDEFGHJKLMNPRSTUVWXY'
CHARS64 = 'HMFLGW5XC301234567899Z67YRT2S8ABCDEFGHJKDVEJ4KQPUALMNPRSTUVWXYNB'

first_char = '1'
i = CHARS32.find(first_char)
sum1 = sum(ord(c) for c in sn) & 31
sum2 = sum(CHARS32.find(c) for c in sn) & 31
ext = first_char + CHARS64[sum1 + i] + CHARS64[sum2 + i]
print(ext)
```

That means that the value after the dash is indeed a kind of checksum

tbnobody, 2024-02-15:
With google image search I found the serial of a HERF-600
A11001YTT-16M
28210000FF5A

The calculation of the checksum works and also the first 4 digits of the converted serial match a 2 channel inverter.
Question is, if there is a mathematical solution to come from 2821 to 1141

Eisenhauer1987, 2024-02-16:
Hi I Have a HERF-1800 and the serialnumber transfer process ist working for me, too. My Serial is: A01002AP4-1T9. At the moment the HERF is detected as: HM-600/700/800-2T. That is a problem because it is a 4 String Modell. How can I access the other Strings?

[image]

Eisenhauer1987, 2024-02-16:
I had make a test with the documented serialnumber ranges:

Class | Models | Serial range
-- | -- | --
HM_1CH | HM-300/350/400-1T | 1121
HM_2CH | HM-600/700/800-2T | 1141
HM_4CH | HM-1000/1200/1500-4T | 1161
HMS_1CH | HMS-300/350/400/450/500-1T | 1124
HMS_1CHv2 | HMS-500-1T v2 | 1125
HMS_2CH | HMS-600/700/800/900/1000-2T | 1144
HMS_4CH | HMS-1600/1800/2000-4T | 1164
HMT_4CH | HMT-1600/1800/2000-4T | 1361
HMT_6CH | HMT-1800/2250-6T

only the "HM_2CH | HM-600/700/800-2T | 1141" is working for me. But with the limitation of two strings and many false values.

tbnobody, 2024-02-16:
> only the "HM_2CH | HM-600/700/800-2T | 1141" is working for me.

That's strange because the first 4 numbers are not sent into the air. They are currently just used to determine the device type. Can you provide a snapshot of the serial/web console? Then we would see what's sent and what's received.

tbnobody, 2024-02-16:
Can you also provide the output if you change the first digits from `1141` to `1161` (for a 4 channel device)?

tbnobody, 2024-02-16:
That's great to hear...
Then I would assume we have to implement the following replacements:
2821 --> 1141
2801 --> 1161

Eisenhauer1987, 2024-02-16:
Yes: 2801 --> 1161 is right for me. Additional a fix in the Webgui is needed, because it is not possible to edit a device with a serial number that include a-f digits

tbnobody, 2024-02-16:
Yes that's clear. I am already working on a solution.

tbnobody, 2024-02-16:
If you want, you can test it. See attached firmware. (Translations are still not implemented)

[opendtu-generic_esp32.zip](https://github.com/tbnobody/OpenDTU/files/14316546/opendtu-generic_esp32.zip)

Birkenstab, 2024-02-18:
> If you want, you can test it. See attached firmware. (Translations are still not implemented)
> 
> [opendtu-generic_esp32.zip](https://github.com/tbnobody/OpenDTU/files/14316546/opendtu-generic_esp32.zip)

It works! Thank you very much!

The only thing that is not working consistently now, is the power limit. I can set it relative but after a restart of opendtu, opendtu show a limit of 0 %. And setting it absolute seems to be not working at all for me. Though I'm not sure if this is related to the Herf inverter at all

Everything else works great :+1: : creating an inverter with herf serial number, setting power limit, power settings

Birkenstab, 2024-02-18:
Now we just need to find out which serial numbers the single channel herf inverters (HERF-300/HERF-400/HERF-500) use?

Birkenstab, 2024-02-18:
In [Discord](https://discord.com/channels/984173303147155506/1146388875204378636/1168303347078541362) stefant posted a regex (I'm not really sure where he got it from). regex: `^((A01)|(A11)|(A21))[A-HJ-NR-Y0-9P]{0,6}-[A-HJ-NP-Z0-9]{3}$`
So if 
A01 (0x2801) => 4 channel => 0x1161
A11 (0x2821) => 2 channel => 0x1141

Then
A21 (0x2841) is probably 1 channel => 0x1121

tbnobody, 2024-02-18:
> opendtu show a limit of 0 %.

The limit is read from the inverter ~10min after a startup of OpenDTU. Otherwise it will lead to many error messages in the event log under some circumstances.
Can you doublecheck this by waiting some time?

Eisenhauer1987, 2024-02-18:
For me the Powerlimt ist working as expected. 0% After Startup from the dtu. After a while, it Displays 77% Like i setted a day before (persistent relative). And I can set all values (nonpersistent relative).

Birkenstab, 2024-02-18:
> The limit is read from the inverter ~10min after a startup of OpenDTU. Otherwise it will lead to many error messages in the event log under some circumstances. Can you doublecheck this by waiting some time?

Ah yes, you are right. After a few minutes the power limit is shown. Then everything should be fine

tbnobody, 2024-02-18:
Could you please doublecheck whether the device info dialogue shows correct values?

Eisenhauer1987, 2024-02-19:
I think you mean this page?!
[image]
Looks fine for me - the production date ist strange, but ......

Also, the Grid information Page is working fine:

[image]

tbnobody, 2024-02-19:
@Birkenstab could you maybe also paste the device info dialog?
@Eisenhauer1987 as you can see, the inverter is currently marked as "Unknown". That means it would be nice to have at least the Harware part numbers of your both inverters to determine a logic of the numbers.

Eisenhauer1987, 2024-02-28:
Little reminder, can somebody else - who using a HERF - paste the device info dialog?

Birkenstab, 2024-02-28:
[image]

matietz, 2024-03-01:
Thanks for the description @Birkenstab. Here is my device info (for a HERF 800):

[image]

maval1ty, 2024-03-03:
Hi there,

I applied the adjusted firmware, entered the S/N of my HERF-800 (A11003G47-131) which is then converted to 11410001c087.

The inverter is added properly (as far as I can tell) but no connection to the inverter is established.

The console doesn't provide any connection related information, here is a snip after a restart:

Admin AP remaining seconds: 10 / 180
09:01:43.204 > Websocket: [/livedata][1] connect
09:01:53.151 > Admin AP remaining seconds: 20 / 180
09:01:57.447 > Websocket: [/livedata][1] disconnect
09:02:03.159 > Admin AP remaining seconds: 30 / 180
09:02:13.170 > Admin AP remaining seconds: 40 / 180
09:02:23.181 > Admin AP remaining seconds: 50 / 180
09:02:33.190 > Admin AP remaining seconds: 60 / 180
09:02:43.207 > Admin AP remaining seconds: 70 / 180
09:02:53.218 > Admin AP remaining seconds: 80 / 180
09:03:03.220 > Admin AP remaining seconds: 90 / 180
09:03:05.344 > Websocket: [/livedata][2] connect
09:03:13.229 > Admin AP remaining seconds: 100 / 180
09:03:23.238 > Admin AP remaining seconds: 110 / 180
09:03:33.255 > Admin AP remaining seconds: 120 / 180
09:03:43.264 > Admin AP remaining seconds: 130 / 180
09:03:53.275 > Admin AP remaining seconds: 140 / 180
09:04:03.278 > Admin AP remaining seconds: 150 / 180
09:04:13.301 > Admin AP remaining seconds: 160 / 180
09:04:23.300 > Admin AP remaining seconds: 170 / 180
09:04:33.314 > Admin AP remaining seconds: 180 / 180
09:04:34.321 > Admin mode disabled
09:08:55.447 > Websocket: [/livedata][2] disconnect

Sorry to bother you with this but am I missing something? Did I do something wrong? Any advice would be highly appreciated.

tbnobody, 2024-03-03:
@maval1ty please provide a screenshot of 
* "Settings" --> "Inverters"
* "Info" --> "System" --> "Radio Information"
* "Info" --> "Ntp"

maval1ty, 2024-03-03:
Thanks for your swift reply!

Sure, here are the screemshots:

[image]

[image]

[image]

So there seems to be an issue with the radio not being configured, right?

I am using this hardware:
https://www.ebay.de/itm/125772523122

tbnobody, 2024-03-03:
As you can see, your RF Module is not configured. Therefor nothing can be send to the inverter. This is something which has to be fixed first. I don't know the pin assignment, therefor I cannot tell you which of the pin mappings you need.

maval1ty, 2024-03-03:
Many thanks for your support!
Once I get this sorted out I will provide you the device information dialog.

robin0793, 2024-03-03:
I got the same issue as maval1ty, my RF Module is configured though. Console log looks the same.
SN is A11003114, converted to 114100018495, but I also get no data

[image]
[image]
[image]
[image]
[image]

I soldered the CMT2300a to a small PCB (with antenna and capacitor) and the ESP32 myself, so I'm not 100% sure there aren't any issues there. But I checked everything and it seems fine, and it says connected in system info

maval1ty, 2024-03-03:
Thanks to your support and the correct PIN-Mapping it is working now!
Here is a screenshot of the device information dialog:

[image]

tbnobody, 2024-03-03:
@robin0793 you have configured a CMT2300a RF module but your inverter requires a NRF24 module

robin0793, 2024-03-03:
> @robin0793 you have configured a CMT2300a RF module but your inverter requires a NRF24 module

Well... I guess that kind of explains it 😬
Thanks!

matietz, 2024-03-04:
> Thanks to your support and the correct PIN-Mapping it is working now! Here is a screenshot of the device information dialog:
> 
> ![grafik](https://private-user-images.githubusercontent.com/162009560/309548767-add29671-7f22-41e9-a2ff-e8bd8129597b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1NDk5OTMsIm5iZiI6MTcwOTU0OTY5MywicGF0aCI6Ii8xNjIwMDk1NjAvMzA5NTQ4NzY3LWFkZDI5NjcxLTdmMjItNDFlOS1hMmZmLWU4YmQ4MTI5NTk3Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNFQxMDU0NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kY2RkMWEwNGMyNmY1NGVmM2VkMjA5OGNlY2RiNmRjZTcxZTkwZTkzY2RjZjQxNjIyYWFlNjdhYzYyYmQ1NjhjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.CA_LbnnS__xoiMa6h15XFypyntaWlLj4SMjAIr8PcCc)

Can you share which pin mapping you used? I'm using the same hardware and somehow lost my mapping after installing the firmware linked here: https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1949420320.

matietz, 2024-03-04:
How is it possible that the efficiency is higher than 100%? This is a screenshot from today using two panels and the HERF 800. When I had only one panel connected the efficiency was even over 300%.
I noticed similar high efficiencies in the opening post ([link](https://private-user-images.githubusercontent.com/4836713/305171325-dad6b22e-49d1-41a4-b36d-92991a09ef6b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1ODIxMTEsIm5iZiI6MTcwOTU4MTgxMSwicGF0aCI6Ii80ODM2NzEzLzMwNTE3MTMyNS1kYWQ2YjIyZS00OWQxLTQxYTQtYjM2ZC05Mjk5MWEwOWVmNmIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMDRUMTk1MDExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTY0YTQ1MGVkOGEyN2RhN2VmM2YxZjRjZmY1NzJjNTNiOGUzYjA3NTFmN2NmNWZmNDkzNzEyNDU4ZWVjYmY5NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.2BTgcSmC_oXCzuozvS9N8v5xdYHv7exDgcq3VMn_7Vo)) and another post in this thread ([link](https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1947832839)). The one person in this thread using a HERF inverter with 4 strings seems to be getting a valid number ([link](https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1947914738)).

[image]

tbnobody, 2024-03-04:
Ok.... If I would guess....
Phase 1 seems to be correct
Temperature looks correct.

String 1: 36,3V x 1,36A = 49,5W
String 2: 32,5V x 1,62A = 52,9W

Then 97,3 / (49,5 + 52,9) = 95,3% Efficiency

That would mean that for the inverters voltage, current and power are a little bit scrambled.

What do you think, are the values for YieldDay and YieldTotal realistic?

tbnobody, 2024-03-04:
Are we able to find someone with a 1 Channel HERF?

tbnobody, 2024-03-04:
The attached file contains a different byte assignment for the 2 channel HERF inverter.
But you have to delete the inverter and re-enter it again (or change your serial number to `282100018d11` (will also be done automatically if you re-add it)
[generic_esp32.zip](https://github.com/tbnobody/OpenDTU/files/14487602/generic_esp32.zip)

matietz, 2024-03-05:
Here is a datapoint from this morning (only one panel attached):
[image]

And here is the accompanying log:
[opendtu_herf800.log](https://github.com/tbnobody/OpenDTU/files/14491666/opendtu_herf800.log)
 
> Ok.... If I would guess.... Phase 1 seems to be correct Temperature looks correct.
>
> String 1: 36,3V x 1,36A = 49,5W String 2: 32,5V x 1,62A = 52,9W
>
> Then 97,3 / (49,5 + 52,9) = 95,3% Efficiency [...]

At first sight they look realistic but I can try to verify this a bit more. 

> The attached file contains a different byte assignment for the 2 channel HERF inverter. But you have to delete the inverter and re-enter it again (or change your serial number to `282100018d11` [...]

Thanks, I will try this tomorrow!

robin0793, 2024-03-05:
There's my device Info using your update from yesterday:
[image]
[image]

Looking good 👍

krk-elektro, 2024-03-06:
> Are we able to find someone with a 1 Channel HERF?

With my next Order in China i can buy one, but this need much Time.
On Stock i have only HERF800 & HERF1600.
With my own HERF1600 all looks good except Inverter informations.
[image]

Eisenhauer1987, 2024-03-06:
Some additional Info for the HERF-1800 and powerlimits:

As far as I know: The Hoymilies Devices limits each output to the configured powerlimit. This means for a 1800w 4-string Hoymilies with confirgured powerlimit of 25% (1800/4*0.25) a max Input of 112W per Panel (DC Input). The Herf handles this different. If you configure 25%, it limits simply the AC Output. So each string can have very different power (max the wp of the Panel) --> Better for East/ West

matietz, 2024-03-06:
> The attached file contains a different byte assignment for the 2 channel HERF inverter. But you have to delete the inverter and re-enter it again (or change your serial number to `282100018d11` [...]

Looking good now, thank you very much 👍 
[image]

Gasperle, 2024-03-11:
Hello! This is my hardware information: 

[image]

Currently using a HERF 800 inverter. It works fine (tried it with the convertion of the serialnumber with the guide and also have the above firmware installed now. Hope it helps, so that it will be fully supported. 

[image]

tbnobody, 2024-03-11:
@Gasperle if you use the .bin file mentioned here: https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1977530583

It should already work. But you have to remove and re-add it again as you've used a 1141 prefix.

Gasperle, 2024-03-11:
> @Gasperle if you use the .bin file mentioned here: [#1751 (comment)](https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1977530583)
> 
> It should already work. But you have to remove and re-add it again as you've used a 1141 prefix.

Thanks for your reply. Is it enough to update to the firmware or do i have to completly reinstall it? 
Because if I add the not converted number (28210001B495) it is showing unkown device.

tbnobody, 2024-03-11:
Updating the firmware is enough. But you really need the .bin file as mentioned in the linked post. (There are two .bin files, you need the most up-to-date ne)
[image]

Gasperle, 2024-03-11:
Got it! Thanks alot! 

[image]

[image]

BigUsh, 2024-03-12:
> Thanks to your support and the correct PIN-Mapping it is working now! Here is a screenshot of the device information dialog: [...]

@maval1ty 
I ordered it from the same dealer and have the same problem as you. Which pin assignment did you use? Or you can link or make them available here.

matietz, 2024-03-12:
> @maval1ty I ordered it from the same dealer and have the same problem as you. Which pin assignment did you use? Or you can link or make them available here.

You should be able to use this one: https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles/nodemcu_esp32.json#L2-L15

BigUsh, 2024-03-12:
> > > Dank Ihrer Unterstützung und dem richtigen PIN-Mapping funktioniert es jetzt! Hier ist ein Screenshot des Geräteinformationsdialogs:
> > > ![Grafik](https://private-user-images.githubusercontent.com/162009560/309548767-add29671-7f22-41e9-a2ff-e8bd8129597b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDQ4ODAsIm5iZiI6MTcxMDI0NDU4MCwicGF0aCI6Ii8xNjIwMDk1NjAvMzA5NTQ4NzY3LWFkZDI5NjcxLTdmMjItNDFlOS1hMmZmLWU4YmQ4MTI5NTk3Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxMlQxMTU2MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZmE0YTg3YTM3ODFiZmZiNGU4ZDNkMDNjY2QzOTI0N2QwNTc5YTdjNDExMDA4YzllMThiM2RlMmM4M2E2ODFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.0yp3WPShsQsgc6GohGC4Wza9GkpXPZIJmjO8gXKNUGE)
> > 
> > 
> > @maval1tyIch habe es beim selben Händler bestellt und habe das gleiche Problem wie Du. Welche Pinbelegung hast du verwendet? Oder Sie können sie hier verlinken oder zur Verfügung stellen.
> 
> Es sollte mit diesem funktionieren: https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles/nodemcu_esp32.json#L2-L15

Thanks
That's exactly what I took.
In the hardware settings I can still select profiles there.
Which one should I take?
No matter which one is selected, the data is always displayed incorrectly

[image]
[image]

gigab0nus, 2024-03-13:
I also got my HERF-1800 running thanks to your guide, so here's my hardware info:
[image]

tbnobody, 2024-03-13:
@gigab0nus dont use any kind of workaround... This will not work anymore as I release a new version. Please just use the version mentioned here: https://github.com/tbnobody/OpenDTU/issues/1751#issuecomment-1977530583
and enter the serial as shown on your inverter!
And a cencored "Hardware-Teilenummer" is useless. But you can try the firmware mentioned above if it's already included in the list of known numbers.

maval1ty, 2024-03-15:
> > > Thanks to your support and the correct PIN-Mapping it is working now! Here is a screenshot of the device information dialog:
> > > ![grafik](https://private-user-images.githubusercontent.com/162009560/309548767-add29671-7f22-41e9-a2ff-e8bd8129597b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDQ4ODAsIm5iZiI6MTcxMDI0NDU4MCwicGF0aCI6Ii8xNjIwMDk1NjAvMzA5NTQ4NzY3LWFkZDI5NjcxLTdmMjItNDFlOS1hMmZmLWU4YmQ4MTI5NTk3Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxMlQxMTU2MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZmE0YTg3YTM3ODFiZmZiNGU4ZDNkMDNjY2QzOTI0N2QwNTc5YTdjNDExMDA4YzllMThiM2RlMmM4M2E2ODFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.0yp3WPShsQsgc6GohGC4Wza9GkpXPZIJmjO8gXKNUGE)
> > 
> > 
> > @maval1ty I ordered it from the same dealer and have the same problem as you. Which pin assignment did you use? Or you can link or make them available here.
> 
> It should work with this one: https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles/nodemcu_esp32.json#L2-L15

@BigUsh
I can confirm that it works well for me with the above linked profile. I just selected the profile "NRF24" and did not change any PIN mapping.

tbnobody, 2024-03-15:
Implemented in f995287a6e721d5be06d9bdedfb354f76be2eaf6

closed 2024-03-15 by tbnobody

BigUsh, 2024-03-17:
Hello,

Unfortunately I still have the problem that the string display doesn't fit.

I have already updated to the current version v24.3.15
I also selected “NRF24” as the hardware profile.

Even after resetting to factory settings, the problem still persists.
Can someone tell me what exactly I'm doing wrong or what else I could test/change?
[image]
[image]
[image]
[image]
[image]
[image]

tbnobody, 2024-03-17:
your serial begins with 1141... that means you used  the workaround in the first post... dont do that. delete the inverter again and type in the serial number as shown on the label of the inverter... I would bet it does not begin with 1141....

BigUsh, 2024-03-19:
> your serial begins with 1141... that means you used the workaround in the first post... dont do that. delete the inverter again and type in the serial number as shown on the label of the inverter... I would bet it does not begin with 1141....

Good bet. What can I say, you're so right.
It works now. Many many thanks.
"If you can read it can be so easy :wink:"

Mihre7, 2024-04-09:
Hello, can someone tell me how and where I can run the script to calculate the serial number?

tbnobody, 2024-04-09:
Not need. Just install the latest openDTU version and insert the version number as shown on the inverter.

[2 more replies]