tbnobody/OpenDTU #1759: [Request] Show OpenDTU as separate PV Charger in the VRM

andreas-bulling, 2024-02-16:
### Describe the solution you'd like

OpenDTU is currently not shown as a separate PV Charger in the Victron VRM - like multiple MPPTs are. It would be nice if OpenDTU was also shown separately to more easily see its yield.

tbnobody, 2024-02-16:
Are you sure that this repo is the right place for this feature request? OpenDTU does not contain any kind of Victron support by itself. Maybe you mean [OpenDTU-on-Battery](https://github.com/helgeerbe/OpenDTU-OnBattery)?

andreas-bulling, 2024-02-17:
yes, I am very sorry - accidentally posted this to the wrong repo.

closed 2024-02-17 by andreas-bulling