tbnobody/OpenDTU #1767: Nach Update < 24.1.26 keine "aktuelle Anzeige"

Fooose, 2024-02-20:
### Is your feature request related to a problem? Please describe.

Moin,
wie der Titel schon sagt funktioniert bei mir mach dem Upgrade die Anzeige nicht mehr.

[image]
[image]
[image]

### Describe the solution you'd like

Vielleicht hat ja jemand einen Tipp, was ich übersehe oder/und falsch gemacht habe.

tbnobody, 2024-02-20:
Sieht so aus als würde dein Browser keine WebSockets unterstützen. Auch bei deinem funktionierenden Bild war der Indikator schon rot. Die neue Version verwendet ausschließlich Websockets zum Update der Daten.

Fooose, 2024-02-20:
Vielen Dank für deine schnelle Antwort, ich habe "Nginx Proxy" im Einsatz und da vergessen die Unterstützung für WebSockets zu aktivieren.

closed 2024-02-20 by Fooose