tbnobody/OpenDTU #1769: ESP32-S3 keine PSRAM Anzeige

Juergen2453, 2024-02-21:
### What happened?

Keine PSRAM Anzeige.
Getestet mit:
ESP32-S3-DevKitC-1 N16R8 (32-Bit-LX7-Dual-Core-Prozessor mit bis zu 240 MHz, 512 KB SRAM, 384 KB ROM, 16 KB RTCSRAM, 8 MB PSRAM)

### To Reproduce Bug

.

### Expected Behavior

.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

v24.2.16

### Anything else?

Einstellung VSC
[image]

Einstellung pin_mapping.json
[image]

Ergebnis
[image]

tbnobody, 2024-02-21:
That is not a problem of opendtu but of the platformio configuration of your board. 
You could e.g. try this: https://github.com/platformio/platform-espressif32/pull/921#issuecomment-1766069497

Juergen2453, 2024-02-21:
Super Danke für den link.
Dieser Eintrag hat geholfen:

[image]

Danach habe ich PSRAM:

[image]

Kann dann geschlossen werden.

closed 2024-02-21 by Juergen2453