tbnobody/OpenDTU #1770: Yield Day is Not correct when inverter got. to „sleep“ after sundown

Elektron79, 2024-02-21:
### What happened?

I habe a lager installation consisting of 6 Inverter.
As expected the yield is increasing oder the day, but as soon as the First Inverter switch off (at sundown) the yield day is decreased and remains unchanged with the Value of the Last Seen Inverter.

### To Reproduce Bug

Connect more than ohne Inverter (at lebst two)
Let the Inverter produce and Monitor the Value of the yield (on the webpage and oder MQTT)
Wait until one Inverter is switched off while sundown.
The value of yield day, is decreased.

### Expected Behavior

The yield day Value is not decreased when a Inverter is switched off / offline.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V24.2.16

tbnobody, 2024-02-21:
Thats a normal behavior. Because it seems that your inverter toggles on/off multiple times in the evening. And on each reset, the values are resetted too. Did you enable "Yield Day correction"? It should elliminate exatcly this.

Elektron79, 2024-02-21:
Thank you for the fast reply.
I changed the setting and will check it tomorrow.

Best regards Michael

Elektron79, 2024-02-26:
After enabling the Feature, it Worms for me.
So would vote to enable this Feature by default.

Best regards and Thank you for all the work!
Michael

closed 2024-02-29 by tbnobody