tbnobody/OpenDTU #1771: DTU hängt sich auf

Marc2805, 2024-02-21:
### What happened?

Ich sende meinen Hoymiles HM-1500 alle 10 Sekunden via IOBroker openDTU adapter das aktuelle Einspeise Limit über den Datenpunkt limit_nonpersistent_absolute um Nulleinspeisung zu realisieren. Problem daran ist das ich die opendtu zwischendurch immer wieder aufhängt (1-2 mal am Tag, zu erkennen ist das an der letzten Aktualisierung teilweise 51354 Sekunden.  ab und zu lässt sich die opendtu über Neustart zurücksetzen, notfalls über Spannungsunterbrechung. 

### To Reproduce Bug

Durch setzen von Power Limit über den Datenpunkt limit_nonpersistent_absolute

### Expected Behavior

funktion ohne absturz

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.2.16

tbnobody, 2024-02-21:
Das ist ein bekanntes Phänomen und kein Issue!
10 Sekunden ist viel zu schnell. Erhöhe die Zeit. In dem Moment wird die Command Queue so lang das kein Platz mehr ist um Daten abzurufen. Steuer Commands haben immer Vorrang vor den Statistik Commands.

Marc2805, 2024-02-21:
ok vielen dank für die schnelle Antwort! Wie weit muss ich die Zeit mindestens erhöhen ?

tbnobody, 2024-02-21:
Das kann ich so direkt nicht beantworten. Einfach aus probieren. In den Disussions gab es da schon ein paar Beiträge. Es wurde damals absichtlich kein minimales Limit eingebaut weil $Menschen selbstständig bis ans Limit gehen wollten.

Marc2805, 2024-02-21:
Vielen dank 
dann versuche ich das mal

closed 2024-02-21 by Marc2805