tbnobody/OpenDTU #1773: New 2,4inch OLED display - flipped/shifted

njordan77, 2024-02-21:
### What happened?

I'm on the latest version of OpenDTU with an ESP32. Used the smaller OLED for some time, never had any issues, now changed to the 2,4inch version. works fine for a day, but every now/then the display flips upside down after starting in the morning, or even more annoying - the display is not flipped but the whole display is shifted in terms of strips of lines are mis-aligned....

not sure if this is well known, im using the I2C variant.

### To Reproduce Bug

its not simple to reproduce as after a new startup it never happens immediately. but almost every day the display is either flipped or shifted in packets of lines and therefore misaligned and unreadable.

### Expected Behavior

_

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.2.16

tbnobody, 2024-02-21:
This is most likely not a issue of OpenDTU but of your wiring. Make sure you have good solder joints, short cables and proper capacitors built in.

njordan77, 2024-02-21:
OK, thanks for the quick reply. soldering wise not sure as i never touch the equipment, so nothing should have the chance to get loose. but i can add capacitors to support the supply voltage to not drop. lets hope this may fix it. so i guess you have not seen this before and is not a well known software issue. Thanks, regards Norbert

tbnobody, 2024-02-21:
> so i guess you have not seen this before and is not a well known software issue. 

No, but I've seen it with my own large oled/lcd displays and a poor power supply (without capacitors) or very long cables.

tbnobody, 2024-03-31:
Closing due to no further activitiy.

closed 2024-03-31 by tbnobody