tbnobody/OpenDTU #1776: MQTT only working after Enable/Disable

Rene1695, 2024-02-23:
### What happened?

Since a couple of days I noticed that the OpenDTU MQTT is not working as in the past. After one of the latest updates I did not see any MQTT messages in the MQTT explorer. 

When I  go to network settings, Disable MQTT and Enable it back again than MQTT is working as expected. But once I gets night and a new day starts, I need to do this work-around again.

### To Reproduce Bug

Wait for a new morning. Go to MQTT Explorer (no MQTT messages, but inverter has already started). Go to network settings - Disable MQTT. Enable MQTT.

### Expected Behavior

It should work without cycling the MQTT setting and this was working in the past versions.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.2.16

### Anything else?

[image]
 
[image]

[image]

tbnobody, 2024-02-23:
What shows the MQTT connection status if MQTT is not working?

Rene1695, 2024-02-23:
@tbnobody  I forgot to track it today. But I will report it tomorrow once the error is back again!

Rene1695, 2024-02-24:
@tbnobody 
1.) Here is the MQTT connection status when MQTT is not working:
[image]

2.) Yes I am sending power limit commands via NodeRED based on my power consumption. I think NodeRED is configured to send this 24/7. Should I configure it to only send this commands when the inverter is online or what do you think could be the problem with this? 

3.) No, the WIFI is configured to run 24/7

Here you can see MQTT Explorer when MQTT is not working:
[image]

And here you can see MQTT Explorer when I reset the MQTT in OpenDTU
[image]

Rene1695, 2024-02-24:
[image]

This one seems to be connected now, after I restarted the MQTT service. Is there some kind of advanced log to see why he did not connect correctly on start-up?

tbnobody, 2024-02-24:
The serial console...
https://www.opendtu.solar/firmware/howto/serial_console/

Rene1695, 2024-02-24:
Okey, I will try it tomorrow. Hopefully it's working for Mac as well.

Rene1695, 2024-02-25:
@tbnobody Wenn ich mich mit der Konsole verbinde bekomme ich diesen Output (MQTT ist dabei wieder wie oben auf Disabled)

`=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2024.02.25 10:46:14 =~=~=~=~=~=~=~=~=~=~=~=
Fetch inverter: 114185074019
TX RealTimeRunData Channel: 3 --> 15 85 07 40 19 80 13 41 40 80 0B 00 65 DB 0C 61 00 00 00 00 00 00 00 00 0C C5 CD 
Interrupt received
RX Channel: 61 --> 95 85 07 40 19 85 07 40 19 01 00 01 01 57 00 4E 01 0C 01 57 00 4E 01 0C 00 02 97 | -80 dBm
Interrupt received
RX Channel: 61 --> 95 85 07 40 19 85 07 40 19 02 16 1B 00 01 FC 23 00 2C 00 2C 08 FD 13 89 02 00 29 | -80 dBm
Interrupt received
RX Channel: 61 --> 95 85 07 40 19 85 07 40 19 83 00 00 00 16 03 E8 00 7C 00 02 8B F8 E6 | -80 dBm
RX Period End
Success
Fetch inverter: 114185074019
TX RealTimeRunData Channel: 23 --> 15 85 07 40 19 80 13 41 40 80 0B 00 65 DB 0C 66 00 00 00 00 00 00 00 00 3C E3 DC 
Interrupt received
RX Channel: 61 --> 95 85 07 40 19 85 07 40 19 01 00 01 01 57 00 4D 01 09 01 57 00 4D 01 09 00 02 97 | -80 dBm
Interrupt received
RX Channel: 61 --> 95 85 07 40 19 85 07 40 19 02 16 1B 00 01 FC 23 00 2C 00 2C 08 FB 13 89 01 FA D6 | -80 dBm
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 83 00 00 00 16 03 E8 00 7E 00 02 61 37 C1 | -80 dBm
RX Period End
Success
Fetch inverter: 114185074019
TX RealTimeRunData Channel: 40 --> 15 85 07 40 19 80 13 41 40 80 0B 00 65 DB 0C 6B 00 00 00 00 00 00 00 00 AC BB 19 
Interrupt received
RX Channel: 75 --> 95 85 07 40 19 85 07 40 19 01 00 01 01 57 00 4D 01 07 01 56 00 4D 01 06 00 02 97 | -80 dBm
Interrupt received
RX Channel: 75 --> 95 85 07 40 19 85 07 40 19 02 16 1B 00 01 FC 23 00 2C 00 2C 09 04 13 89 01 F5 27 | -80 dBm
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 83 00 00 00 16 03 E8 00 7E 00 02 AF D3 EB | -80 dBm
RX Period End
Success
Fetch inverter: 114185074019
TX RealTimeRunData Channel: 61 --> 15 85 07 40 19 80 13 41 40 80 0B 00 65 DB 0C 70 00 00 00 00 00 00 00 00 5C 05 4C 
Interrupt received
RX Channel: 75 --> 95 85 07 40 19 85 07 40 19 01 00 01 01 57 00 4C 01 04 01 57 00 4C 01 03 00 02 90 | -80 dBm
Interrupt received
RX Channel: 75 --> 95 85 07 40 19 85 07 40 19 02 16 1B 00 01 FC 23 00 2C 00 2C 09 03 13 88 01 EF 3B | -80 dBm
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 83 00 00 00 15 03 E8 00 7E 00 02 A2 87 B1 | -80 dBm
RX Period End
Success
Fetch inverter: 114185074019
TX RealTimeRunData Channel: 75 --> 15 85 07 40 19 80 13 41 40 80 0B 00 65 DB 0C 75 00 00 00 00 00 00 00 00 0C 3A 26 
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 01 00 01 01 56 00 4B 01 01 01 56 00 4B 01 00 00 02 96 | -80 dBm
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 83 00 00 00 15 03 E8 00 7B 00 02 F2 9A F9 | -80 dBm
RX Period End
Middle missing
Request retransmit: 2
TX RequestFrame Channel: 3 --> 15 85 07 40 19 80 13 41 40 82 DE 
RX Period End
Middle missing
Request retransmit: 2
TX RequestFrame Channel: 23 --> 15 85 07 40 19 80 13 41 40 82 DE 
Interrupt received
RX Channel: 23 --> 95 85 07 40 19 85 07 40 19 02 16 1B 00 01 FC 23 00 2C 00 2C 09 03 13 88 01 EA 3E | -80 dBm
RX Period End
Success
Fetch inverter: 114185074019
TX RealTimeRunData Channel: 40 --> 15 85 07 40 19 80 13 41 40 80 0B 00 65 DB 0C 7A 00 00 00 00 00 00 00 00 FC 7B 98 
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 01 00 01 01 56 00 4A 00 FE 01 56 00 4A 00 FD 00 02 94 | -80 dBm
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 02 16 1B 00 01 FC 23 00 2C 00 2C 09 05 13 88 01 E5 37 | -80 dBm
Interrupt received
RX Channel: 3 --> 95 85 07 40 19 85 07 40 19 83 00 00 00 15 03 E8 00 7A 00 02 5E EB 25 | -80 dBm
RX Period End
Success
Fetch inverter: 114185074019
TX RealTimeRunData Channel: 61 --> 15 85 07 40 19 80 13 41 40 80 0B 00 65 DB 0C 7F 00 00 00 00 00 00 00 00 AC 44 F2 
Interrupt received
RX Channel: 23 --> 95 85 07 40 19 85 07 40 19 01 00 01 01 56 00 4A 00 FC 01 56 00 49 00 FB 00 02 93 | -80 dBm
Interrupt received
RX Channel: 23 --> 95 85 07 40 19 85 07 40 19 83 00 00 00 15 03 E8 00 7B 00 02 6B 05 FF | -80 dBm
RX Period End
Middle missing
Request retransmit: 2
TX RequestFrame Channel: 75 --> 15 85 07 40 19 80 13 41 40 82 DE 
Interrupt received
RX Channel: 23 --> 95 85 07 40 19 85 07 40 19 02 16 1B 00 01 FC 23 00 2C 00 2C 09 04 13 88 01 E1 32 | -80 dBm
RX Period End
Success
`

Ich hatte vorhin auch das Problem, dass der NTP sich nicht synchronisiert hat. Habe diesen jetzt mal wie vorgeschlagen auf meine Fritz!Box gelegt, denke das sollte helfen.

Trotzdem bleibt das Problem, dass ich immer MQTT einmal manuell aus- und wieder einschalten muss, damit es läuft. Hast du noch irgendwelche Ideen dazu? Danke :)

tbnobody, 2024-02-25:
> 2.) Yes I am sending power limit commands via NodeRED based on my power consumption. I think NodeRED is configured to send this 24/7. Should I configure it to only send this commands when the inverter is online or what do you think could be the problem with this?

How often do you  send these commands? And do you also send them at night when the inverter is not reachable? Both can lead to a memory exhaustion.

Rene1695, 2024-02-25:
I send it every 5 seconds. Because I can read my current meter, I try to avoid feed in energy to the public net. And yes, I also send it nights (was easier to set up in NodeRED). 

I thought the 5 second intervall is already the limitation to prevent the MCU from memory exhaustion? At least I will try to stop sending the commands at night. This setup was working for weeks. It's not likely that something in this direction has changed in some of the newer version right?

tbnobody, 2024-02-25:
5 seconds is a very short inverval. There are threads where people tested it and even 10 seconds was too short. At night it gets even worth because if the inverter is not reachable it takes longer to send commands (because have to wait for a timeout) and the command queue fills up.

Rene1695, 2024-02-27:
Thanks for the info! I will try to completely disable sending the limit commands just for testing. Yesterday I set up node red to only send the inverter limit from 8-16. Today I noticed that the inverter started at 7:18 but there was the yellow tag mentioned "last communication yesterday 17:45" and no data was present in Open DTU. I waited until now (there is enough light so there should be some power output) but I needed to restart OpenDTU. After restart, I could see the data and communication was back again. But still I needed to restart MQTT services manually because status was again on "not connected".

Mhm is it possible to roll back to an older version? I would like to see if this is related to the software version.

closed 2024-03-01 by Rene1695