tbnobody/OpenDTU #1783: Inverter poolt Nachts

Juergen2453, 2024-02-24:
### What happened?

Ich frage drei Inverter ab HM-600, HM-400, HMS1600. Über Tag läuft alles normal aber Nachts pollt der HMS weiter.
Alle drei haben die selbe Einstellung.
[image]

[image]

### To Reproduce Bug

.

### Expected Behavior

Sollte der HMS nicht auch aufhören zu pollen.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

v24.2.16

tbnobody, 2024-02-24:
In dem Moment ist der Inverter offline, du hast aber das senden der Befehle noch aktiv. Damit Befehle gesendet werden können muss eine Verbindung zum Inverter bestehen. Daher wird versucht die Frequenz des Inverters zu setzen (das muss immer vor jeglicher Kommunikation passieren) was auch mal gut und gerne 15min dauern kann, selbst wenn er erreichbar ist. Die Software kann ja nicht ahnen das kein Befehl gesendet werden wird.
Also schalte das senden der Befehle bei Nacht aus und es wird nichts mehr gesendet.

Juergen2453, 2024-02-24:
"Befehle auch nachts senden" habe ich ausgeschaltet (nur bei dem HMS) und er sendet nicht mehr.
Bei den HM ist es anscheinend nicht nötig "Befehle auch nachts senden" auszuschalten?

tbnobody, 2024-02-24:
> Bei den HM ist es anscheinend nicht nötig "Befehle auch nachts senden" auszuschalten?

Korrekt. Nur bei HMS/HMT Invertern muss man die Frequenz anpassen. HM Inverter arbeiten auf fest definierten Frequenzen (bis auf das Frequenz Hopping) und müssen nicht eingestellt werden.

Juergen2453, 2024-02-24:
Danke und ich Schließe

closed 2024-02-24 by Juergen2453