tbnobody/OpenDTU #1787: [Request] turn off automatic check of update of firmware

4c2, 2024-02-25:
### Is your feature request related to a problem? Please describe.

I turned on the automatic check for update under: Info -> System -> Firmware update. 
Later on I wanted to turn it of. But unfortunately, there is no option to turn it off again. 

### Describe the solution you'd like

have an option under Firmware update to turn off the automated check (again). 

### Describe alternatives you've considered

1) leave it as it is: disadvantage: github is contacted every log in. it is privacy protection issue "only"
2) add "check once" button: may cause additional work and benefit is less

4c2, 2024-02-27:
yes, this works. In firefox, one can find the settings here: 

[image]
(set from 1 --> 0 and refresh page. 

Shall we leave the issue open for further improvements?

MetaChuh, 2024-02-27:
> Shall we leave the issue open for further improvements?

probably yes, as it sounds reasonable to be able to reset this value from within the opendtu gui

closed 2024-03-15 by tbnobody

4c2, 2024-03-17:
@tbnobody also thanks from my side. It works well in v24.3.15.

commit bd8d93bf (2024-03-15): Feature: Allow enabling and disabling of the version check

Fix #1787