tbnobody/OpenDTU #1797: Total Yield Day is partly not visible when value is > 10.0000Wh.

4c2, 2024-02-27:
### What happened?

Total Yield Day is partly not visible when value is > 10.0000Wh. 
When screensaver moves the text to the left, the text disappears. 

("Heute" in following pics is the Total Yield Day.)

[image]
[image]

### To Reproduce Bug

Use 1.3" OLED display with fusion board. 
German language, but maybe issue also for others. 
have day yield of > 10.000Wh
When screensaver moves the text to the left, the text disappears. 

### Expected Behavior

Total Yield Day always visible. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.2.16

### Relevant log/trace output

```Shell
no log available
```

### Anything else?

-

broth-itk, 2024-02-28:
I wish I had the same problem ;-)

Maybe it might make sense to convert down to kWh above 9999 Wh -> 10.0 kWh
Total length of the string would fit.

tm-107, 2024-02-29:
Same problem here... today I'm currently at >26,000 Wh and the value disappears when it's on the left side.

The problem only exists if the display is set to German.
Changing the display language to English or French solves the problem.
In German, either the font would have to be a little smaller or - as already suggested - there should be an automatic switch from Wh to kWh.

MetaChuh, 2024-02-29:
@tm-107 👍🙏

> The problem only exists if the display is set to German.

thanks for fiddling around with other languages, until you found a possible cause 👍

from a perspective of string lengths, it's very intreaguing why `today:` works, while `Heute:` does not, as the string lenghts should be the same.

maybe the difference lies at the capital `H`which consumes more pixel width ?
i've not worked very much with the display libraries yet, but i, probably same as the project owner and chief of dev @tbnobody assumed, that the additional pixels would just not be displayed, if the image is shifted ... but not that the whole line would be ommited, as it seems the case

very fascinating.
can anyone else with a larger than 10kwh daily solar yield confirm this as well ?

tbnobody, 2024-02-29:
I could imagine that either this line: https://github.com/tbnobody/OpenDTU/blob/43c3c8d30c935169327d7d2de686093df26e85a2/src/Display_Graphic.cpp#L120
or this line 
https://github.com/tbnobody/OpenDTU/blob/43c3c8d30c935169327d7d2de686093df26e85a2/src/Display_Graphic.cpp#L117
calculates a negative result which results in a underflow and therefor in a very high number (e.g. 240) and places the text outside the display.

tm-107, 2024-03-01:
Here is a video-shot of the problem. Maybe it's interesting to see at which positions the text disappears.
[Youtube-Link](https://youtube.com/shorts/bR0NKePMGzs?feature=share)

closed 2024-03-15 by tbnobody

broth-itk, 2024-03-17:
@MetaChuh Thanks for your wishes! ;-)
Have a great day!

4c2, 2024-03-21:
@tbnobody thanks also from my side. It works with v24.3.15.

tm-107, 2024-03-23:
> let's hope for a good solar yield today too. and especially my wishes for @broth-itk to surpass a daily yield to something like `22222 Wh` ... with the fun part, that exactly this number might break the pixel length fix again 😉 (just kidding, in theory the fix should work for any pixel width)

Unfortunately, the display will not show “22222 Wh” because it now automatically switches to kWh.

[image]

commit 50abcd10 (2024-03-15): Fix: Prevent hiding text on display if it's too long

Fixes: #1797