tbnobody/OpenDTU #1798: [Request] Add description for error 152

phol, 2024-02-29:
### Is your feature request related to a problem? Please describe.

My inverters showed error code 152 on HMT-2250-6T, which is listed as "Unknown" in OpenDTU and not included in the Hoymiles manual. 
Therefore, I reached out to Hoymiles support.

### Describe the solution you'd like

This is the answer I received from Hoymiles support

"The 152 code is for the AC failure. When the phase angle difference between any two phases of the ABC phase exceeds 5° and the count exceeds 10 times, the alarming show on the platform."

Could you please add this description for error code 152.

phol, 2024-02-29:
I think something like this should be added to `src/parser/AlarmLogParser.cpp`, but I'm getting error `too many initializer values`, so therefore I chose to open an issue, rather then a merge request, as I don't have any experience with C++.

```cpp
    { AlarmMessageType_t::ALL, 152, "AC failure: The phase angle difference between any two phases exceeded 5° >10 times", "", "" },
```

homeautomation2022, 2024-03-02:
Change https://github.com/tbnobody/OpenDTU/blob/master/lib/Hoymiles/src/parser/AlarmLogParser.h#L11 to 131
and add
`{ AlarmMessageType_t::HMT, 152, "AC failure: The phase angle difference between any two phases exceeded 5° >10 times", "", "" },` here: https://github.com/tbnobody/OpenDTU/blob/master/lib/Hoymiles/src/parser/AlarmLogParser.cpp#L63

@phol Can you please thank Hoymiles and ask them via email whether you can get a current list of all error codes, as unfortunately some are missing from the instructions. We've been waiting for the missing ones for a long time, that would be nice.

tbnobody, 2024-03-02:
I've already included it in my local development branch. Will be part of the next release

lumapu, 2024-03-03:
would be nice to have all the error / warning codes. Does anybody searched already in the Hoymiles Cloud? I think they should be somewhere listed as JSON. Maybe it helps again to decompile a recent version of the Hoymiles App @DanielR92

closed 2024-03-15 by tbnobody

phol, 2024-03-20:
@homeautomation2022 I have requested a full list of codes, but unfortunately, Hoymiles customer service is not allowed to share it with customers. Therefore, I think we'll indeed have to rely on decompiling apps, searching through the cloud and asking whenever they pop up, as I just did in #1847 .

commit 021d9b5f (2024-03-15): Feature: Added  description for alarm id 152

Fixes: #1798