tbnobody/OpenDTU #1800: Add Dates in event log in Live-View

martinstejskal, 2024-03-01:
### Is your feature request related to a problem? Please describe.

There are only timestamps in event log. It is hard to understand what event was raised on what day. In my setup I use batteries, so my inverter runs 24/7.

### Describe the solution you'd like

Add the date additionally to the timestamp in event log.

### Describe alternatives you've considered

Alternatively the date could be shown on mouseover as a tooltip.

### Additional context

This feature request increases the meaningfulness of the event log, by adding needed information of the date of an event.

tbnobody, 2024-03-01:
The eventlog is reset at each reboot of the inverter. Therefor you can 100% assume that  it's only the current day. 
On the other hand, in the event view os only shown what the inverter deliveres.

tbnobody, 2024-03-01:
But I have no chance to know on which day the event occoured. If you e.g. don't reset the inverter over the night, the inverter returns something like 37:00:00.... Now I know that the time is 37:00:00 - 24:00:00 = 1 day 13:00....
But I don't know which date was 1 day... because it depends on the startup date of the inverter (which is not known).

phol, 2024-03-01:
Ok that makes sense, as it's the exact same issue I'm facing when trying to parse the websocket data and removing duplicate events. 
Perhaps adding a separate date field, indicating on which date the OpenDTU received the event from the receiver would be an option? Or adding some logic to remove duplicates?

tbnobody, 2024-03-01:
The problem is here, that the inverter sends ALL events. I request the events and get a response with the last 15 events. The inverter itself remove some events which seem to be less important if an more then 15 events occour on one day.

phol, 2024-03-28:
To (at least partially) solve this problem, I've tried to analyse it and have made a proposal / small feature request in #1875 .