tbnobody/OpenDTU #1805: Hardware-Model unknown

svalcken, 2024-03-03:
### What happened?

The Hardware Model is not detected.
Model No: 124097
Model Hoymiles HMS-800-T2
[image]

### To Reproduce Bug

Nothing to do. Automatic detected.

### Expected Behavior

Right detection of inverter.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V24.2.16

svalcken, 2024-03-03:
[image]

tbnobody, 2024-03-03:
could you please doublecheck after a inverter restart (so most likley tomorrow) and a dtu restart? the hardware part number looks too short.

tbnobody, 2024-03-03:
Because of #645, #1066, #1204, #1234, #1250 and #1254 which all reported the same Hardware-Part-Number I will define `124097` as invalid and trigger a re-request of the data.

svalcken, 2024-03-03:
Ist mir heute aufgefallen, weil ich mich gewundert habe, dass der Wechselrichter mit mehr rund 611 statt Max 600w mir angezeigt wurde.
Darauf habe ich gesehen, dass das Grid Profil falsch ist und sehr kurze Hex String angezeigt wurde.

Und danach sah ich, dass der Wechselrichter nicht richtig angezeigt wurde.

Ich teste morgen, ob der Wechselrichter und das Grid wieder richtig erkannt wird.
Wechselrichter ist jetzt aus.
OpenDTU habe ich gerade rebootet.

Melde mich dann nochmal.

svalcken, 2024-03-04:
Heute Morgen wurde der Wechselrichter korrekt erkannt und die Modelnummer ist auch länger.

Gleiches gilt fürs Gridprofil, dass gestern zu kurz war.

Kann es sein, dass der Speicher vollgelaufen ist oder dass die Übertragung nicht vollständig war?

Es muss ja ein Grund geben, warum die Daten falsch waren.

homeautomation2022, 2024-03-04:
Übertragungsproblem ist am wahrscheinlichsten und anschließend falsch geparste Daten aus den Bytes.

tbnobody, 2024-03-04:
> Übertragungsproblem ist am wahrscheinlichsten 

Wäre aber krasser zufall. weil die ganzen CRC prüfsummen passen ja. ansonsten würden die pakete ja sofort verworfen. irgendwas verhaspelt da der inverter. wie gesagt. die 124097 nehme ich zum anlass einen re-request zu senden. mal sehen was so passiert :)

closed 2024-03-09 by svalcken