tbnobody/OpenDTU #1814: add AC yieldtotal in HA like sensor and enable expiration for a part of sensor [Request]

smiley-2, 2024-03-06:
### Describe the solution you'd like

Hello,
sorry for my english, I can't find the ac yieldtotal for all of my device in HA, only for each one.
An I think, if it's possible, It's great to enable the expiration but not for the ac yieldday total and for each device.
the value reset when we pass midnight, it's possible?

thanks!

hschurer, 2024-03-08:
There is a MQTT topic that stores this value, but it's no automatically visible in Home-Assistant.
AC Yieldtotal of all inverters is available in a MQTT Topic : opendtu/ac/yieldtotal or solar/ac/yieldtotal depending on your naming convention.
You can add this sensor to HA bij manually listen on MQTT on this topic.

Regarding the value resets you can set these parameters individually on each inverter in their settings on OpenDTU
[image]

smiley-2, 2024-03-09:
a big thanks for your help!

MetaChuh, 2024-03-09:
@smiley-2 
please close this issue, if the info from @hschurer has solved this issue for you.

greetings and thanks,
metachuh

closed 2024-03-09 by smiley-2