tbnobody/OpenDTU #1822: Yield Day is Not correct when inverter got. to „sleep“ after sundown

Elektron79, 2024-03-10:
### What happened?

This is Not the Same as #1770, Bit it‘s close to.
Berauschend of that i opened a new issue.

i noticed tonight that (After Setting the "Yield Day correction") the yield day Value was Not reseted for all Inverter (Types?)
In my Installation I have HMS-2000-4T and HMS-450-1T.
As you can See in the Screenshots, the yield oder the HMS-2000-4T was reseted to „0“ After Midnight, but for the HMS-450-1T it was Not done.

As the yield of the HMS-450-1T is Not increasing infinitely, it must ne reseted at some Point in Time, but obviously Not at Midnight.

### To Reproduce Bug

Let the Inverter produce oder time, wait for Midnight and Check the yielt day Value.

### Expected Behavior

All Inverter Seouls behave the Same.
[image]
[image]

the Screenshot where cremtest shortly After Midnight.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V24.2.16

njordan77, 2024-03-14:
i have the same issue, but there is a setting that stops resetting of the inverter restarts in the evening due to short.term light conditions that make it restart and start from Zero again.

aapee73, 2024-06-16:
i have the same issue also. If i reboot opendtu then the "Total Yield Day" will show the correct value. I have three Hoymiles HMT-2250 in my installation.

closed 2025-04-18 by github-actions[bot]