tbnobody/OpenDTU #1835: Prometheus metric opendtu_YieldTotal and opendtu_YieldDay are missing

wesseli01, 2024-03-16:
### What happened?

Among the prometheus metrics exposed by OpenDTU, there are also yield metrics total and per day. The sum fields are missing since last update.

### To Reproduce Bug

Open http://hostname.of.opendtu/api/prometheus/metrics
Search for # TYPE opendtu_YieldTotal
Search for # TYPE opendtu_YieldDay

### Expected Behavior

opendtu_YieldTotal and opendtu_YieldDay should be included again.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.15

### Relevant log/trace output

```Shell
# HELP opendtu_build Build info
# TYPE opendtu_build gauge
opendtu_build{name="OpenDTU-48FB10",id="v24.3.15",version="0.1.27"} 1
# HELP opendtu_platform Platform info
# TYPE opendtu_platform gauge
opendtu_platform{arch="ESP32-D0WD-V3",mac="A8:42:E3:48:FB:10"} 1
# HELP opendtu_uptime Uptime in seconds
# TYPE opendtu_uptime counter
opendtu_uptime 22186
# HELP opendtu_heap_size System memory size
# TYPE opendtu_heap_size gauge
opendtu_heap_size 285868
# HELP opendtu_free_heap_size System free memory
# TYPE opendtu_free_heap_size gauge
opendtu_free_heap_size 74956
# HELP opendtu_biggest_heap_block Biggest free heap block
# TYPE opendtu_biggest_heap_block gauge
opendtu_biggest_heap_block 36852
# HELP opendtu_heap_min_free Minimum free memory since boot
# TYPE opendtu_heap_min_free gauge
opendtu_heap_min_free 44848
# HELP wifi_rssi WiFi RSSI
# TYPE wifi_rssi gauge
wifi_rssi -54
# HELP wifi_station WiFi Station info
# TYPE wifi_station gauge
wifi_station{bssid="3C:37:12:9C:98:BF"} 1
# HELP opendtu_last_update last update from inverter in s
# TYPE opendtu_last_update gauge
opendtu_last_update{serial="114191236269",unit="0",name="HM1"} 15462
# HELP opendtu_inverter_limit_relative current relative limit of the inverter
# TYPE opendtu_inverter_limit_relative gauge
opendtu_inverter_limit_relative{serial="114191236269",unit="0",name="HM1"} 1.000000
# HELP opendtu_inverter_limit_absolute current relative limit of the inverter
# TYPE opendtu_inverter_limit_absolute gauge
opendtu_inverter_limit_absolute{serial="114191236269",unit="0",name="HM1"} 800.000000
# HELP opendtu_Power in W
# TYPE opendtu_Power gauge
opendtu_Power{serial="114191236269",unit="0",name="HM1",type="AC",channel="0"} 0.0
# HELP opendtu_Voltage in V
# TYPE opendtu_Voltage gauge
opendtu_Voltage{serial="114191236269",unit="0",name="HM1",type="AC",channel="0"} 235.4
# HELP opendtu_Current in A
# TYPE opendtu_Current gauge
opendtu_Current{serial="114191236269",unit="0",name="HM1",type="AC",channel="0"} 0.00
# HELP opendtu_Frequency in Hz
# TYPE opendtu_Frequency gauge
opendtu_Frequency{serial="114191236269",unit="0",name="HM1",type="AC",channel="0"} 50.00
# HELP opendtu_PowerFactor in 
# TYPE opendtu_PowerFactor gauge
opendtu_PowerFactor{serial="114191236269",unit="0",name="HM1",type="AC",channel="0"} 0.000
# HELP opendtu_ReactivePower in var
# TYPE opendtu_ReactivePower gauge
opendtu_ReactivePower{serial="114191236269",unit="0",name="HM1",type="AC",channel="0"} 0.0
# HELP opendtu_PanelInfo panel information
# TYPE opendtu_PanelInfo gauge
opendtu_PanelInfo{serial="114191236269",unit="0",name="HM1",channel="0",panelname="HM1_1"} 1
# HELP opendtu_MaxPower panel maximum output power
# TYPE opendtu_MaxPower gauge
opendtu_MaxPower{serial="114191236269",unit="0",name="HM1",channel="0"} 410
# HELP opendtu_YieldTotalOffset panel yield offset (for used inverters)
# TYPE opendtu_YieldTotalOffset gauge
opendtu_YieldTotalOffset{serial="114191236269",unit="0",name="HM1",channel="0"} 0.000000
opendtu_Power{serial="114191236269",unit="0",name="HM1",type="DC",channel="0"} 0.5
opendtu_Voltage{serial="114191236269",unit="0",name="HM1",type="DC",channel="0"} 20.3
opendtu_Current{serial="114191236269",unit="0",name="HM1",type="DC",channel="0"} 0.02
opendtu_YieldDay{serial="114191236269",unit="0",name="HM1",type="DC",channel="0"} 353
opendtu_YieldTotal{serial="114191236269",unit="0",name="HM1",type="DC",channel="0"} 97.599
opendtu_Irradiation{serial="114191236269",unit="0",name="HM1",type="DC",channel="0"} 0.122
opendtu_PanelInfo{serial="114191236269",unit="0",name="HM1",channel="1",panelname="HM1_2"} 1
opendtu_MaxPower{serial="114191236269",unit="0",name="HM1",channel="1"} 410
opendtu_YieldTotalOffset{serial="114191236269",unit="0",name="HM1",channel="1"} 0.000000
opendtu_Power{serial="114191236269",unit="0",name="HM1",type="DC",channel="1"} 0.6
opendtu_Voltage{serial="114191236269",unit="0",name="HM1",type="DC",channel="1"} 20.3
opendtu_Current{serial="114191236269",unit="0",name="HM1",type="DC",channel="1"} 0.03
opendtu_YieldDay{serial="114191236269",unit="0",name="HM1",type="DC",channel="1"} 405
opendtu_YieldTotal{serial="114191236269",unit="0",name="HM1",type="DC",channel="1"} 115.240
opendtu_Irradiation{serial="114191236269",unit="0",name="HM1",type="DC",channel="1"} 0.146
opendtu_PowerDC{serial="114191236269",unit="0",name="HM1",type="INV",channel="0"} 1.1
opendtu_YieldDay{serial="114191236269",unit="0",name="HM1",type="INV",channel="0"} 758
opendtu_YieldTotal{serial="114191236269",unit="0",name="HM1",type="INV",channel="0"} 212.839
opendtu_Temperature{serial="114191236269",unit="0",name="HM1",type="INV",channel="0"} 11.2
opendtu_Efficiency{serial="114191236269",unit="0",name="HM1",type="INV",channel="0"} 0.000
opendtu_last_update{serial="114184623107",unit="1",name="HM2"} 15667
opendtu_inverter_limit_relative{serial="114184623107",unit="1",name="HM2"} 1.000000
opendtu_inverter_limit_absolute{serial="114184623107",unit="1",name="HM2"} 800.000000
opendtu_Power{serial="114184623107",unit="1",name="HM2",type="AC",channel="0"} 0.0
opendtu_Voltage{serial="114184623107",unit="1",name="HM2",type="AC",channel="0"} 236.0
opendtu_Current{serial="114184623107",unit="1",name="HM2",type="AC",channel="0"} 0.00
opendtu_Frequency{serial="114184623107",unit="1",name="HM2",type="AC",channel="0"} 49.97
opendtu_PowerFactor{serial="114184623107",unit="1",name="HM2",type="AC",channel="0"} 0.000
opendtu_ReactivePower{serial="114184623107",unit="1",name="HM2",type="AC",channel="0"} 0.0
opendtu_PanelInfo{serial="114184623107",unit="1",name="HM2",channel="0",panelname="HM2_1"} 1
opendtu_MaxPower{serial="114184623107",unit="1",name="HM2",channel="0"} 425
opendtu_YieldTotalOffset{serial="114184623107",unit="1",name="HM2",channel="0"} 0.000000
opendtu_Power{serial="114184623107",unit="1",name="HM2",type="DC",channel="0"} 0.6
opendtu_Voltage{serial="114184623107",unit="1",name="HM2",type="DC",channel="0"} 16.3
opendtu_Current{serial="114184623107",unit="1",name="HM2",type="DC",channel="0"} 0.03
opendtu_YieldDay{serial="114184623107",unit="1",name="HM2",type="DC",channel="0"} 391
opendtu_YieldTotal{serial="114184623107",unit="1",name="HM2",type="DC",channel="0"} 102.267
opendtu_Irradiation{serial="114184623107",unit="1",name="HM2",type="DC",channel="0"} 0.141
opendtu_PanelInfo{serial="114184623107",unit="1",name="HM2",channel="1",panelname="HM2_2"} 1
opendtu_MaxPower{serial="114184623107",unit="1",name="HM2",channel="1"} 425
opendtu_YieldTotalOffset{serial="114184623107",unit="1",name="HM2",channel="1"} 0.000000
opendtu_Power{serial="114184623107",unit="1",name="HM2",type="DC",channel="1"} 0.6
opendtu_Voltage{serial="114184623107",unit="1",name="HM2",type="DC",channel="1"} 16.3
opendtu_Current{serial="114184623107",unit="1",name="HM2",type="DC",channel="1"} 0.03
opendtu_YieldDay{serial="114184623107",unit="1",name="HM2",type="DC",channel="1"} 380
opendtu_YieldTotal{serial="114184623107",unit="1",name="HM2",type="DC",channel="1"} 94.922
opendtu_Irradiation{serial="114184623107",unit="1",name="HM2",type="DC",channel="1"} 0.141
opendtu_PowerDC{serial="114184623107",unit="1",name="HM2",type="INV",channel="0"} 1.2
opendtu_YieldDay{serial="114184623107",unit="1",name="HM2",type="INV",channel="0"} 771
opendtu_YieldTotal{serial="114184623107",unit="1",name="HM2",type="INV",channel="0"} 197.189
opendtu_Temperature{serial="114184623107",unit="1",name="HM2",type="INV",channel="0"} 11.2
opendtu_Efficiency{serial="114184623107",unit="1",name="HM2",type="INV",channel="0"} 0.000
```

wesseli01, 2024-03-16:
# HELP opendtu_YieldDay in Wh
# HELP opendtu_YieldTotal in kWh

tbnobody, 2024-03-16:
I don't understand... I see a lot of `opendtu_YieldTotal` and `opendtu_YieldDay`... They are available for each string and per inverter...

wesseli01, 2024-03-16:
Please look, the sum fields for both are missing since last update. They are not available! Only the fields per inverter in that.

http://hostname/api/livedata/status

{"inverters":[{"serial":"114191236269","name":"HM1","order":0,"data_age":6610,"poll_enabled":true,"reachable":false,"producing":false,"limit_relative":100,"limit_absolute":800},{"serial":"114184623107","name":"HM2","order":1,"data_age":6405,"poll_enabled":true,"reachable":false,"producing":false,"limit_relative":100,"limit_absolute":800}],"total":{"Power":{"v":0,"u":"W","d":1},"YieldDay":{"v":1529,"u":"Wh","d":0},"YieldTotal":{"v":410.0279846,"u":"kWh","d":3}},"hints":{"time_sync":false,"radio_problem":false,"default_password":false}}

tbnobody, 2024-03-16:
You are showing me JSON output of the web interface. Are we talking about the JSON output or about the prometheus output?

In either way this was not the last update.... This was the update with version [v24.2.12](https://github.com/tbnobody/OpenDTU/releases/tag/v24.2.12)

Please read the changelog especially the Breaking Changes!

The AC items where moved to the INV section because on inverters with more then one Phase the Yield values are not available per phase but only for the whole inverter.

wesseli01, 2024-03-16:
I have sent the Prometheus output at first. 

I have updated from v24.2.12 to current version. 

Which fields should I take for the both sum fields from the output now?[https://github.com/tbnobody/OpenDTU/issues/1835#issue-2190216381](url)

tbnobody, 2024-03-16:
opendtu_YieldDay{serial="114184623107",unit="1",name="HM2",type="INV",channel="0"} 771
opendtu_YieldTotal{serial="114184623107",unit="1",name="HM2",type="INV",channel="0"} 197.189

wesseli01, 2024-03-16:
It’s wrong. The value of YiieldDay is 410.028 and for YielTotal is 1529. I have sent the value. You told me only the value of HM2. I need the sum of HM1 and HM2.

I can find the values in the own field over the WEB too.

tbnobody, 2024-03-16:
> I can find the values in the own field over the WEB too.

But they did never appear in the prometheus API.

wesseli01, 2024-03-16:
Perfect, I'm sorry. The solution is very simple. Tx.

closed 2024-03-16 by wesseli01