tbnobody/OpenDTU #1836: Not able to update from v23.10 to higher one

mcfuhrt, 2024-03-17:
### What happened?

I cannot update from version v23.10.9 to a higher version. I even get an OTA error for the next version.
[image]
[image]

### To Reproduce Bug

Downgrade to v23.10.9 and try to upgrade to on of the newer versions.

### Expected Behavior

Successful update to any of the next versions

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.10.9

tbnobody, 2024-03-17:
Please search for existing issues before opening a new one. Your one is related to #1501
Provide a screenshot of Info --> System --> Memory Information
If the max sketch size is not equal 1920 please follow this manual:  https://www.opendtu.solar/firmware/howto/upgrade_partition/

tbnobody, 2024-03-31:
Closed due to further activity. Feel free to reopen or create a new issue if the problem persists.

closed 2024-03-31 by tbnobody