tbnobody/OpenDTU #1846: After updating the DTU Last Night, today the First Inverter Shows no Details (Type, Gridcode etc.)

Elektron79, 2024-03-20:
### What happened?

I updated my DTU yesterday Night to 24.3.15.
Today I noticed that the Details of the First Inverter was missing.
The Hardware Typ Number was 124097 which was Not correct.
Also the Gridcode was not shown correct.

I have a Multi Inverter Setup. Ohne of the HMS2000-4t was not recognized correct, the other wäre correct.

After a reboot everything was fine.

### To Reproduce Bug

Update the DTU and wait Till the Inverter restart.

### Expected Behavior

The Value should be correct Even After an Update.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.3.15

tbnobody, 2024-03-20:
>  Also the Gridcode was not shown correct.

What do you mean with grid code? The Gridprofile itself?

> The Hardware Typ Number was 124097 which was Not correct.

I already made an attempt to exclude the part number 124097 
https://github.com/tbnobody/OpenDTU/blob/f66b4fa5f115453d9863d917fd40f32116b9a30e/lib/Hoymiles/src/parser/DevInfoParser.cpp#L207

But the logic is wrong. Should be unequal instead of equal.

It is not related to an update of OpenDTU but just of internal issue of the inverter itself. It sends this crappy data with a valid checksum etc. Will fix the workaround soon.

Elektron79, 2024-03-20:
Hi, 
Sorry, I was not precise enough,
I mean (as you already thought) the Gridprofile. Instead of “DE - DE_VDE4105_2018” the Gridprofile “US - NA_IEEE1547_240V” was shown.

Also the Inverter Hardware_Type number should be “270692630” instead of “124097”.

But if this is caused by the Inverter, and after a restart of OpenDTU the problem was gone, the DTU was able to read out the correct data after some time. I don’t know if the Inverter is broadcasting the Information automatically or if the DTU has to ask for the information. Maybe DTU can ask again for the information after some minutes?

best regards Michael

tbnobody, 2024-03-20:
> Maybe DTU can ask again for the information after some minutes?

That was the intention of the patch above (which I  implemented wrong). The DTU has to ask the inverter. 

Regarding the grid profile.... If this error occours again (which will maybe not happen very often) could you tell me the received data? (Should be shown on the very bottom of the dialog). It would be helpfull to know how many bytes are received in this error state.

closed 2024-03-22 by tbnobody

Elektron79, 2024-03-26:
How often is the grid code read out?
Everyday when the Inverter starts or only when the openDTU connects for the first time (after restart) to the Inverter?
Depending on that I have to restart the openDTU every night to trigger the read out…

Best regards Michael

commit 3125f16d (2024-03-22): Fix: Previously check for HwPartNumber 124097 was implemented wrong

Fix: #1846