tbnobody/OpenDTU #1850: [Request] Web-UI: Live view shows all modules in one view

4c2, 2024-03-21:
### Is your feature request related to a problem? Please describe.

I recently had 2 modules with strongly dropped yield. It took me certain days to see this, since they are just from one inverter out of many. 
Moreover, it would be generally nice to have an overview about at least the power and daily of all modules. This is e.g. offered by the original product. (you even have an installation view)

### Describe the solution you'd like

Add installation view where one can view all modules at one view. Similar to original product: 
[image]

alternatively: 
have something like a table view to show all modules:
[image]

in both cases, it would be great to have a selection of what to show. Like: Power, daily yield, ...

### Describe alternatives you've considered

just leave it as it is.

tm-107, 2024-04-24:
Yes that would be great.
I did it via Grafana, but within OpenDTU would be great for debugging.

bierschneider, 2024-07-04:
Hey, otherwise it would be helpful to be able to show live data of more than one inverter in the live view.
Possibly the buttons on the left can be used to toggle data view of one inverter. The data of each selected inverter can be grouped in the main frame. This would help to figure out technical issues faster.