tbnobody/OpenDTU #1851: Data reported from Hoymiles Micro Inverter HMT-1800-6T 3F partially bad

dpiller, 2024-03-21:
### What happened?

I receive data a lot of time, which is quite impossible. Microinverter is Hoymiles Micro Inverter HMT-1800-6T 3F. At the end of the day the summary yields are good, the daily and total too. See attached an example for good and bad data received. 

[image]
[image]

### To Reproduce Bug

Not applicable

### Expected Behavior

Receive consisted data. Or ignore broken-invalid data packets. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v4.4.6-dirty / 2024.03.07 / config 0.1.27 / master

tbnobody, 2024-03-21:
Are you using multiple DTUs in parallel? E.g. the original Hoymiles DTU?
Please provide a console log when bad data is recieved. otherwise it's not possible to check where it's coming from.

dpiller, 2024-03-21:
Only openDTU, this one. 

Fetch inverter: yyyyyyyyyyyy
16:24:29.814 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 2E 00 00 00 00 00 00 00 00 30 96 B8 
16:24:30.000 > RX Period End
16:24:30.205 > All missing
16:24:30.413 > Nothing received, resend whole request
16:24:30.541 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 2E 00 00 00 00 00 00 00 00 30 96 B8 
16:24:30.811 > RX Period End
16:24:30.923 > All missing
16:24:31.125 > Nothing received, resend whole request
16:24:31.434 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 2E 00 00 00 00 00 00 00 00 30 96 B8 
16:24:31.637 > RX Period End
16:24:31.875 > All missing
16:24:31.957 > Nothing received, resend whole request
16:24:32.262 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 2E 00 00 00 00 00 00 00 00 30 96 B8 
16:24:32.353 > RX Period End
16:24:32.456 > All missing
16:24:32.564 > Nothing received, resend whole request
16:24:32.904 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 2E 00 00 00 00 00 00 00 00 30 96 B8 
16:24:33.080 > RX Period End
16:24:33.179 > All missing
16:24:33.411 > Nothing received, resend count exeeded
16:24:33.583 > TX AlarmData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 11 00 65 FC 51 2E 00 00 00 00 00 00 00 00 EA 8D 63 
16:24:33.796 > RX Period End
16:24:33.889 > All missing
16:24:33.999 > Nothing received, resend whole request
16:24:34.198 > TX AlarmData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 11 00 65 FC 51 2E 00 00 00 00 00 00 00 00 EA 8D 63 
16:24:34.403 > RX Period End
16:24:34.538 > All missing
16:24:34.709 > Nothing received, resend whole request
16:24:34.918 > TX AlarmData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 11 00 65 FC 51 2E 00 00 00 00 00 00 00 00 EA 8D 63 
16:24:35.125 > Fetch inverter: xxxxxxxxxxxx
16:24:35.354 > TX RealTimeRunData Channel: 61 --> 15 83 06 35 12 80 12 88 76 80 0B 00 65 FC 51 33 00 00 00 00 00 00 00 00 60 03 C8 
16:24:35.527 > Interrupt received
16:24:35.835 > RX Channel: 23 --> 95 83 06 35 12 83 06 35 12 01 00 01 01 1E 00 46 00 41 00 C9 00 BA 00 02 2F 6E BD | -80 dBm
16:24:35.941 > Interrupt received
16:24:36.155 > RX Channel: 3 --> 95 83 06 35 12 83 06 35 12 02 00 02 17 00 02 5E 02 45 01 1D 00 42 00 45 00 BB 39 | -80 dBm
16:24:36.346 > Interrupt received
16:24:36.550 > RX Channel: 3 --> 95 83 06 35 12 83 06 35 12 03 00 C5 00 02 14 D0 00 02 34 11 02 40 02 6A 08 6E FE | -80 dBm
16:24:36.897 > Interrupt received
16:24:37.167 > RX Channel: 75 --> 95 83 06 35 12 83 06 35 12 84 13 82 02 DC 00 B6 00 22 03 CB 00 F2 00 07 25 6E BC | -80 dBm
16:24:37.373 > RX Period End
16:24:37.573 > All missing
16:24:37.880 > Nothing received, resend whole request
16:24:38.088 > TX AlarmData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 11 00 65 FC 51 2E 00 00 00 00 00 00 00 00 EA 8D 63 
16:24:38.193 > RX Period End
16:24:38.430 > Success
16:24:38.595 > RX Period End
16:24:38.803 > All missing
16:24:39.356 > Nothing received, resend whole request
16:24:39.967 > TX AlarmData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 11 00 65 FC 51 2E 00 00 00 00 00 00 00 00 EA 8D 63 
16:24:40.136 > RX Period End
16:24:40.335 > All missing
16:24:40.549 > Nothing received, resend count exeeded
16:24:40.648 > Fetch inverter: yyyyyyyyyyyy
16:24:40.849 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 38 00 00 00 00 00 00 00 00 50 70 28 
16:24:41.053 > RX Period End
16:24:41.468 > All missing
16:24:41.561 > Nothing received, resend whole request
16:24:41.972 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 38 00 00 00 00 00 00 00 00 50 70 28 
16:24:42.178 > RX Period End
16:24:42.382 > All missing
16:24:42.587 > Nothing received, resend whole request
16:24:42.792 > TX RealTimeRunData 864.50 MHz --> 15 65 20 07 48 80 12 88 76 80 0B 00 65 FC 51 38 00 00 00 00 00 00 00 00 50 70 28 
16:24:42.996 > Interrupt received
16:24:43.099 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 01 00 01 01 53 00 01 00 05 00 03 00 12 00 02 27 35 A4 | -92 dBm
16:24:43.206 > Interrupt received
16:24:43.405 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 02 00 02 29 32 01 F0 01 E6 01 53 00 01 00 02 00 04 AB | -92 dBm
16:24:43.541 > Interrupt received
[... 19 lines cut ...]
16:24:48.118 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 04 00 05 B0 02 07 0C 2E 12 2E 12 FF FF FF FB B0 02 FD | -92 dBm
16:24:48.329 > Interrupt received
16:24:48.522 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 05 07 17 2E 3F 2E 3F 00 00 00 05 B0 02 07 18 2E 77 17 | -92 dBm
16:24:48.726 > Interrupt received
16:24:49.035 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 06 2E 77 FF FF FF FA B0 02 07 43 2E E9 2E E9 00 00 5F | -92 dBm
16:24:49.238 > Interrupt received
16:24:49.302 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 07 00 06 B0 02 07 46 2E F9 2E F9 FF FF FF FA B0 02 B6 | -92 dBm
16:24:49.354 > Interrupt received
16:24:49.549 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 08 07 47 2E FE 2E FE 00 00 00 05 B0 02 07 48 2F 0D 61 | -92 dBm
16:24:49.750 > Interrupt received
16:24:49.954 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 09 2F 0D FF FF FF FB B0 02 07 49 2F 11 2F 11 00 00 20 | -92 dBm
16:24:50.197 > Interrupt received
16:24:50.364 > RX 864.50 MHz --> 95 65 20 07 48 80 12 88 76 0A 00 06 B0 02 07 4A 2F 3F 2F 3F FF FF FF FA B0 02 B7 | -92 dBm
16:24:50.569 > Interrupt received
[... cut ...]
16:24:52.413 > Success
16:24:52.540 > Fetch inverter: xxxxxxxxxxxx
16:24:52.553 > TX RealTimeRunData Channel: 75 --> 15 83 06 35 12 80 12 88 76 80 0B 00 65 FC 51 3D 00 00 00 00 00 00 00 00 00 4F EA 
16:24:52.820 > Interrupt received
16:24:53.157 > RX Channel: 23 --> 95 83 06 35 12 83 06 35 12 02 00 02 17 00 02 5E 02 45 01 1D 00 42 00 45 00 BB 39 | -80 dBm
16:24:53.332 > Interrupt received
16:24:53.540 > RX Channel: 3 --> 95 83 06 35 12 83 06 35 12 03 00 C5 00 02 14 D1 00 02 34 11 02 41 02 6A 08 6D FD | -80 dBm

tbnobody, 2024-03-21:
> Or ignore broken-invalid data packets.

Packages are only used if they have to correct checksums (yes, there are multiple checksums).

Will doublecheck later if your bad values are coming directly from the inverter. If so nothing can be done because in this case all the checksums are correct and the inverter sends crap with correct checksums.

tbnobody, 2024-03-25:
Your logs contained 8 vaild packages (successfull received + right inverter)
7 packages contained a wattage of zero (also power factor = zero)

[image]

Excel file used for analysis: [GH_Issue_1851.xlsx](https://github.com/tbnobody/OpenDTU/files/14748237/GH_Issue_1851.xlsx)

So unfortunatly the values are coming directly from the inverter. To be honest, I have no clue whats going on.

dpiller, 2024-03-25:
Hmm…and receiving error code 152 unkown a lot of times as well

tbnobody, 2024-03-25:
152 was added in one of the previous releases

https://github.com/tbnobody/OpenDTU/blob/3b05f447d5c2c9e386a432e0b015ca11db2c8595/lib/Hoymiles/src/parser/AlarmLogParser.cpp#L63

closed 2025-04-18 by github-actions[bot]