tbnobody/OpenDTU #1855: After the latest update the details of the inverter and modules are not shown

Disane87, 2024-03-23:
### What happened?

In latest releases the details of the modules and inverter were shown underneath the tiles of the current data. But these are missing now:
[image]

### To Reproduce Bug

Update to the latest release and open the website. 

### Expected Behavior

The detail data should be visible again.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.22

### Anything else?

I already tested other browsers with other devices (pc with edge (and in private tab) against iOS safari). Always the same

tbnobody, 2024-03-23:
You have to refresh the webinterface bei either pressing ctrl+f5 or cleaning the cache.

Disane87, 2024-03-23:
> You have to refresh the webinterface bei either pressing ctrl+f5 or cleaning the cache.

Already did that

closed 2024-03-23 by Disane87