tbnobody/OpenDTU #1856: [Request] publish RF rssi in db

tikismoke, 2024-03-23:
### Is your feature request related to a problem? Please describe.

It would be nice to get RF rssi in db (not sure of the right name).
It could help to see if the position (distance) of the card it good an close enough from the installation.

### Describe the solution you'd like

A value send to ha (through mqtt in dtu status).

tikismoke, 2024-03-23:
Yes this is for wifi rssi.

I'd like the 868Mhz same value

tikismoke, 2024-03-23:
> understood, you want to see the cmt2300a received signal strength indication, not the wifi rssi.
> sorry for misunderstanding, we'll have a look at it, if documentation or features are missing.

Thanks. I do not know if it is even possible of course.

In fact I have a lot of communication drop between cmt and solar parts.
I need to find a good place to let the opendtu card. But without value it's hard to compare best place.

tikismoke, 2024-03-23:
> thx for the info 👍
> 
> btw:
> we are an international community, so please don't hesitate to write your native language thoughts and wishes below your english posts as an addition.

D'accord mais pas sûr que ce soit bénéfique 🤣.

Alright, but not sure on benefits 🤣.

SuperNinja73, 2024-03-25:
je suis d'accord avec @tikismoke 

> I need to find a good place to let the opendtu card. But without value it's hard to compare best place.

car j'ai les mêmes soucis de :
 ```
11:53:47.436 > TX RealTimeRunData 865.00 MHz --> 15 85 00 12 12 80 15 12 28 80 0B 00 66 01 57 BA 00 00 00 00 00 00 00 00 42 D4 A8 
11:53:47.989 > RX Period End
11:53:47.989 > All missing
11:53:47.989 > Nothing received, resend whole request
```

Comme je l'ai deja remonté ici : https://github.com/tbnobody/OpenDTU/issues/980#issuecomment-1977613886, j'ai des probleme de retour de l'inverter HMS1000 (via CMT2300A )  et l'ideal serait d'avoir le RSSI de la liaison que le CMT2300A est capable de mesurer (et non pas le RSSI du WIFI) : https://github.com/DanielR92/CMT2300A/blob/main/EBYTE/AN144-CMT2300AW_RSSI_Usage_Guideline_EN_V0.9.pdf
et nos confreres de chez AHOY : 
https://github.com/lumapu/ahoy/wiki/RF-Module-CMT2300A---E49-900M20S
Est ce qu'on peux espérer un ajout de cette remontée de la force du signal de reception de l'inverter @MetaChuh  ?

tikismoke, 2024-03-25:
> je suis d'accord avec @tikismoke
>
>
>
>
> car j'ai les mêmes soucis de :
>  ```
> 11:53:47.436 > TX RealTimeRunData 865.00 MHz --> 15 85 00 12 12 80 15 12 28 80 0B 00 66 01 57 BA 00 00 00 00 00 00 00 00 42 [...]

Mince pourtant j'ai cherché avant d'ouvrir ce ticket. Bon ben au moins je suis pas le seul qui voudrait bien monitorer cette mesure 👍

SuperNinja73, 2024-03-25:
> @SuperNinja73 ne sait pas. est-ce que ça marche déjà avec ahoy ? pas observé.

Je n'ai pas essayé ahoy,  je me base sur des documents fournis et autres notes d'application. Espérant qu'une âme charitable puisse intégrer cette mesure du RSSI à openDTU 😁

tikismoke, 2024-03-25:
Ceci dit j'ai trouvé un emplacement qui va bien avec une jolie courbe de production/restitution grâce a homeassistant et un shelly en plus du linky(ce con même avec la teleinfo sait pas faire dans le négatif 😹 une honte😔).
Bref c'est vrai qu'après ça sert plus a grand chose mais sur le principe ça serait un +.
C'est tout ce qui m'a manqué en plus de deviner qu'il fallait passer en 868Mhz à 16dB pour que tout fonctionne(dès que je change en + ou en - la fréquence ou le gain ça marche plus du tout) et j'en suis content.

Encore un beau projet libre qui mérite 👍.

tbnobody, 2024-10-05:
Implemented in edfe06e31eae286fbb88da69e9b65420cfa8628a

closed 2024-10-05 by tbnobody