tbnobody/OpenDTU #1857: [Request]: Add Peak Power per Day Attribute

cbouser, 2024-03-23:
### Is your feature request related to a problem? Please describe.

I would like to know & store the peak power per day. Right now, I can query openDTU very frequently via API call and store the max per day. But it would be much more convenient (and come with lower load) if openDTU stored the maximum power (total, per string) on a daily basis and I can fetch it via API . 

### Describe the solution you'd like

Would be great if OpenDTU - during each call - compares the currently reported power (total, per string) against a locally stored maximum and updates the maximum, if needed. Either every night at 0:00 or when the inverter starts for the first time on a day, these values are reset to 0.
These values can be added to the UI and provided via API. For me, it is interesting to see how close the system gets every day vs theoretical peak power - either just for curiosity or to improve placement of panels etc.

tbnobody, 2024-03-24:
> this is the issue section, which is supposed to handle issues regarding the firmware.

That's ok because there is a special feature request form:
[image]

But you are right with the different restrictions regarding the max value.

broth-itk, 2024-03-25:
> if for example the esp32 reboots due to overload watchdog triggers, or if a user performs a firmware update during the day, which will zero the values.

This is an issue for not only these values.

My 2 cents:

I came across this problem with my own products. I started to use the buffered RTC SRAM but this was far from good.
Eventually I added a small I2C FRAM memory to the layout to store my variable data. 

My proposal is to require an external nonvolatile memory with many W/E cycles to keep persistent runtime data across reboots.

cbouser, 2024-03-25:
For me, all the mentioned restrictions/limitations would be fine and acceptable. This is no nuclear power plant controller, one or the other inaccuracy is ok for me- having this feature without 100% correct data under all conditions would still be a significant improvement for me: Main "concern" if I got it right are reboots of the ESP throughout the day (e.g. power outage, watchdog, firmware update). For my use case, that would even almost not matter at all since I query the data every 15 minutes and store it in a "reliable" database - so in case the peak is 400w and after a reboot late during the day, the new peak is only 100W, I would still have stored the 400w and only potentially lost another peak that came after the 400w of last fetch and a reboot. Compared to today where I can only get the max out of the 15 min queries, a huge improvement of data quality. If the ESP doesn't reboot unexpectedly (never happened to me so far) with even higher precision.

Thanks, Christian

phol, 2024-03-26:
If you'd like to get peak power per day values, you could also try out my [OpenDTU Logger](https://git.hollander.online/energy/opendtu-logger). See also my [blog post](https://pieterhollander.nl/post/opendtu-logger/). The included Grafana dashboards can show peak power per inverter, panel or everything added up and contain visualisations for basically any metric possible to retrieve from OpenDTU.