tbnobody/OpenDTU #1865: [Request] Set inverter limit on a pool of two inverters

gitisgreat2023, 2024-03-25:
### Is your feature request related to a problem? Please describe.

When using two Hoymiles inverters simultaneously, it would great to set the power limit to those two inverters as a whole.

### Describe the solution you'd like

Three modes of power limiting:
1) proportional mode. A target of 250 W for a HM-300 and a HM-600 together is achieved by setting 100*250/(300+600)=27% for both HMs. Simplest to implement. For multiple inverters with the same conditions (orientation, shades) this is sufficient.
2) block mode. A target of 250 W for a HM-300 and a HM-600 together is achieved by setting 250/600=42% for the HM-600, the HM-300 is set to 0%. 650W is then achieved with 100% for the HM-600 and 17% for the HM-300. This mode is ideal for OpenDTU-onBattery.
3) adaptive mode. If both inverters produce currently in total more than the overall limit, limit proportionally. If less, set the limit proportionally higher for each inverter, max'ed by 100%. E.g. target is 600, HM-300 produces 275, HM-600 285. Limit for HM-300 is set to 300, HM-600 as well. This is best suited for maxing out on the allowed limit of 600/800 with inverters with different orientations/conditions.

### Describe alternatives you've considered

Using multiple DTUs.

### Additional context

See [this OpenDTU](https://github.com/tbnobody/OpenDTU/discussions/1608) and [this OpenDTU-onBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/issues/230) discussions.

darkazar, 2024-03-25:
It would be great to set inverter limit on a pool of all inverters connected to opendtu. I have 10 to control :) 👍

gitisgreat2023, 2024-03-25:
> It would be great to set inverter limit on a pool of all inverters connected to opendtu. I have 10 to control :) 👍

Yeah... I agree... however the biggest improvement is from 1 to 2... if that works probably with increased polling (5 sec per inverter) it can scale to 10...