tbnobody/OpenDTU #1871: time out

AzraelsDisk, 2024-03-27:
### What happened?

Since the last update of the current version, I am experiencing the problem that the communication between DTU and WR is timed out? often and for a long time.

Currently since 475 seconds... still growing.

No changes in infrastructure or ownership of the DTU. 

### To Reproduce Bug

upgrade to latest version

### Expected Behavior

stable connection as before

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	v24.3.22

### Relevant log/trace output

```Shell
RX Period End
16:34:40.771 > All missing
16:34:40.771 > Nothing received, resend whole request
16:34:40.771 > TX ActivePowerControl Channel: 3 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:42.817 > RX Period End
16:34:42.817 > All missing
16:34:42.817 > Nothing received, resend whole request
16:34:42.817 > TX ActivePowerControl Channel: 23 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:44.841 > RX Period End
16:34:44.841 > All missing
16:34:44.841 > Nothing received, resend whole request
16:34:44.841 > TX ActivePowerControl Channel: 40 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:45.608 > Received MQTT message on topic: solar/116184891783/cmd/limit_nonpersistent_absolute
16:34:45.608 > Limit Non-Persistent: 40 W
16:34:45.754 > Received MQTT message on topic: solar/116184891783/cmd/limit_nonpersistent_absolute
16:34:45.754 > Limit Non-Persistent: 40 W
16:34:46.632 > Received MQTT message on topic: solar/116184891783/cmd/limit_nonpersistent_absolute
16:34:46.632 > Limit Non-Persistent: 40 W
16:34:46.877 > RX Period End
16:34:46.877 > All missing
16:34:46.877 > Nothing received, resend whole request
16:34:46.877 > TX ActivePowerControl Channel: 61 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:48.903 > RX Period End
16:34:48.903 > All missing
16:34:48.903 > Nothing received, resend count exeeded
16:34:48.975 > TX ActivePowerControl Channel: 75 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:50.980 > RX Period End
16:34:50.980 > All missing
16:34:50.980 > Nothing received, resend whole request
16:34:50.980 > TX ActivePowerControl Channel: 3 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:53.011 > RX Period End
16:34:53.011 > All missing
16:34:53.011 > Nothing received, resend whole request
16:34:53.011 > TX ActivePowerControl Channel: 23 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:54.619 > Received MQTT message on topic: solar/116184891783/cmd/limit_nonpersistent_absolute
16:34:54.619 > Limit Non-Persistent: 40 W
16:34:55.056 > RX Period End
16:34:55.056 > All missing
16:34:55.056 > Nothing received, resend whole request
16:34:55.056 > TX ActivePowerControl Channel: 40 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:57.097 > RX Period End
16:34:57.097 > All missing
16:34:57.097 > Nothing received, resend whole request
16:34:57.097 > TX ActivePowerControl Channel: 61 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:34:59.085 > RX Period End
16:34:59.085 > All missing
16:34:59.085 > Nothing received, resend count exeeded
16:34:59.158 > TX ActivePowerControl Channel: 75 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:35:01.212 > RX Period End
16:35:01.212 > All missing
16:35:01.212 > Nothing received, resend whole request
16:35:01.212 > TX ActivePowerControl Channel: 3 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:35:03.231 > RX Period End
16:35:03.231 > All missing
16:35:03.231 > Nothing received, resend whole request
16:35:03.231 > TX ActivePowerControl Channel: 23 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82 
16:35:03.637 > Received MQTT message on topic: solar/116184891783/cmd/limit_nonpersistent_absolute
16:35:03.637 > Limit Non-Persistent: 40 W
16:35:05.264 > RX Period End
16:35:05.264 > All missing
16:35:05.264 > Nothing received, resend whole request
16:35:05.264 > TX ActivePowerControl Channel: 40 --> 51 84 89 17 83 80 15 07 72 81 0B 00 01 90 00 00 B1 00 82
```

tbnobody, 2024-03-27:
> 16:34:45.608 > Received MQTT message on topic: solar/116184891783/cmd/limit_nonpersistent_absolute
> 16:34:45.608 > Limit Non-Persistent: 40 W
> 16:34:45.754 > Received MQTT message on topic: [...]

As you can see, you are setting the powerlimit 3 times in one second each time you do this, a package has to be transmitted to the inverter. This takes some time. During this  time, the inverter has no  chance to send a `RealTimeRunData` command to the inverter to receive current statistics data (as you can see, it does not appear in your log)

It takes even longer if the inverter is not reachable and the package has to be retransmitted. 

So to debug this issue, you should look for a `RealTimeRunData` command and to it's received payload. If it does not appear, you are sending to many control commands.

AzraelsDisk, 2024-03-27:
> It takes even longer if the inverter is not reachable and the package has to be retransmitted.
> 
> So to debug this issue, you should look for a `RealTimeRunData` command and to it's received payload. If it does not appear, you are sending to many control commands.

Thanks for the quick replay
Understoud I have updated it to 15 sec and will comeback if it still persists

tbnobody, 2024-04-29:
I assume the problem did not persist. I am closing this issue. Feel free to reopen or create a new one if it does persist.

closed 2024-04-29 by tbnobody