tbnobody/OpenDTU #1876: [WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"

schnaggelz, 2024-03-30:
### What happened?

The web UI is no longer reachable (timeout) but I can ping the device.

The serial terminal output says e.g.:

```
[46218457][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"
[46218483][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"
[46218515][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"
[46218546][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"
TX ActivePowerControl Channel: 3 --> 51 90 15 51 04 80 13 81 04 81 0B 00 00 64 00 00 BF 40 86 
Interrupt received
```

### To Reproduce Bug

This is happening in the night when the batteries get empty and therefore the inverters go offline.

### Expected Behavior

DTU stays connected to WiFi. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.22

### Relevant log/trace output

```Shell
RX Period End
Success
[46950148][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"
TX ActivePowerControl Channel: 23 --> 51 90 15 51 04 80 13 81 04 81 0B 00 02 58 00 00 0B 81 CD 
Interrupt received
RX Channel: 40 --> D1 90 15 51 04 90 15 51 04 81 00 00 0B 00 14 07 48 | -80 dBm
RX Period End
Success
TX ActivePowerControl Channel: 40 --> 51 90 15 53 22 80 13 81 04 81 0B 00 02 58 00 00 0B 81 E9 
RX Period End
All missing
Nothing received, resend whole request
TX ActivePowerControl Channel: 61 --> 51 90 15 53 22 80 13 81 04 81 0B 00 02 58 00 00 0B 81 E9 
Call to /api/livedata/status temporarely out of resources. Reason: "std::bad_alloc".
RX Period End
All missing
Nothing received, resend whole request
TX ActivePowerControl Channel: 75 --> 51 90 15 53 22 80 13 81 04 81 0B 00 02 58 00 00 0B
```

### Anything else?

See also https://forum.arduino.cc/t/wificlient-cpp-517-flush-fail-on-fd-48-errno-11-no-more-processes-what-it-mean/1098254. I'll try to find a relevant place in the code, but I need to get somehow familiar with it first.

tbnobody, 2024-03-30:
Seems you are running out of resources. This happens if you are sending commands to often while the inverter is not available.

Please also see https://github.com/tbnobody/OpenDTU/issues/1871#issuecomment-2023116725

schnaggelz, 2024-03-30:
Thank you very much for your response! I need to look at my controller code, maybe I don't check the connection states.

tbnobody, 2024-04-12:
Going to close this issue due to further activity. Please feel free to reopen or creating a new one if the problem persists.

closed 2024-04-12 by tbnobody