tbnobody/OpenDTU #1877: pool.ntp.org must not be used by software

bohni, 2024-03-30:
### What happened?

Default NTP server is pool.ntp.org

https://github.com/tbnobody/OpenDTU/blob/3b05f447d5c2c9e386a432e0b015ca11db2c8595/include/defaults.h#L25

### To Reproduce Bug

Fresh install an check NTP config

### Expected Behavior

Us a vendor specific NTP server.

see https://www.ntppool.org/de/vendors.html

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

3b05f44

bohni, 2024-03-30:
So, if you quote, so please quote complete:

Open Source projects are of course particularly welcome to use the pool in their default setup, __but we ask that you [get a vendor zone](https://www.ntppool.org/de/vendors.html#vendor-zone)__ when using the pool as a default configuration.

tbnobody, 2024-03-30:
Thanks for the hint. I am already in contact with them.

bohni, 2024-03-30:
> Thanks for the hint. I am already in contact with them.

You're welcome

commit 6f3b8fb8 (2024-03-31): Fix: Change default NTP server

Fixes #1877
closed 2024-03-31 by tbnobody