tbnobody/OpenDTU #1878: Prometheus API labels changed, breaking dashboards

CommanderRedYT, 2024-03-31:
### What happened?

Instead of using "AC" in "type" labels, it is now "INV". It also feels like some other things have changed, as it does not work 100% correct when replacing "AC" with "INV".

### To Reproduce Bug

Update to [v24.3.22](https://github.com/tbnobody/OpenDTU/releases/tag/v24.3.22)

### Expected Behavior

Database API should be backwards compatible

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.22

tbnobody, 2024-03-31:
Unfortunatly not possible to keep all old api endpoints on a small device like the ESP. Please update your querys or use a previous version.

closed 2024-03-31 by tbnobody

CommanderRedYT, 2024-03-31:
How would you recommend making it so both old and new versions work? I would need that for my app

tbnobody, 2024-03-31:
dont know anything about prometheus... but what about querying for AC.type + INV.type
the first one is filled the other one for new data.