tbnobody/OpenDTU #1888: HM-1500 with Firmware V01.00.30 after uograde from 00.18

yzffrog, 2024-04-05:
### What happened?

after the firmware update at the inverter every read request from the openDTU sets the actual earning value to 0 (resets the value)

### To Reproduce Bug

every read request

### Expected Behavior

as usual - not setting the value to 0

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	v24.3.31

tbnobody, 2024-04-05:
> after the firmware update at the inverter every read request from the openDTU sets the actual earning value to 0 (resets the value)

I don't know what you are meaning. How did you do the firmware update of the inverter? Are you using two DTUs (original + DIY) at the same time? 
Did you do the firmware update at night? The inverter is not reachable at night. Therefor no values can be read.

yzffrog, 2024-04-05:
> I don't know what you are meaning. How did you do the firmware update of the inverter? Are you using two DTUs (original + DIY) at the same time? Did you do the firmware update at night? The inverter [...]

The  firmware update on the inverter was done by HM OTA - cause heve been unspecific HW-Errors i reported to HM

tbnobody, 2024-04-05:
> every read request from the openDTU sets the actual earning value to 0 (resets the value)

if every read request set's it to zero, how do you know that it is unequal zero? If it is set to zero it has to be greater then zero at a specific time. Please explain a little bit more detailed.

And also provide a console log with the transmitted data to doublecheck what's delivered from the inverter towards the DTU.

yzffrog, 2024-04-05:
> if every read request set's it to zero, how do you know that it is unequal zero? If it is set to zero it has to be greater then zero at a specific time. Please explain a little bit more detailed. [...]

I played around with request intervals so i can see date that is cleared after the next request

yzffrog, 2024-04-06:
that how the graph looks like - the field 'Tagesertrag'iss set to 0 on everey request from openDTU
[image]

tbnobody, 2024-04-06:
* Please provide a console log (as text, no screenshot)
* Please provide screenshot of the Device Info Dialog

yzffrog, 2024-04-06:
<html><body>

    
    <div id="app" data-v-app=""><nav class="navbar navbar-expand-md fixed-top bg-body-tertiary" data-bs-theme="dark" style=""><div class="container-fluid"><a href="http://192.168.88.26/" class="navbar-brand" style="display: flex; height: 30px; padding: 0px;"><svg width="30" height="30" viewBox="0 0 16 16" fill="currentColor" role="img" focusable="false" class="d-inline-block align-text-top text-warning"></svg></a></div></nav></div><a href="http://192.168.88.26/" class="navbar-brand" style="display: flex; height: 30px; padding: 0px;"><span style="margin-left: 0.5rem;"> OpenDTU </span></a><div class="collapse navbar-collapse" id="navbarNavAltMarkup"><ul class="navbar-nav navbar-nav-scroll d-flex me-auto flex-sm-fill"><li class="nav-item"><a href="http://192.168.88.26/" class="nav-link">Live View</a></li><li class="nav-item dropdown"><a class="nav-link dropdown-toggle" href="http://192.168.88.26/settings/device#" id="navbarScrollingDropdown" role="button" data-bs-toggle="dropdown" aria-expanded="false">Settings</a></li><li class="nav-item dropdown"><a class="nav-link dropdown-toggle" href="http://192.168.88.26/settings/device#" id="navbarScrollingDropdown" role="button" data-bs-toggle="dropdown" aria-expanded="false">Info</a></li><li class="nav-item"><a href="http://192.168.88.26/about" class="nav-link">About</a></li><li class="flex-sm-fill"></li><li class="nav-item dropdown me-2">*</li></ul></div><li class="nav-item dropdown me-2"></li><form class="d-flex" role="search"></form><nav class="navbar navbar-expand-md fixed-top bg-body-tertiary" data-bs-theme="dark" style=""><div class="container-fluid"><div class="collapse navbar-collapse" id="navbarNavAltMarkup"><ul class="navbar-nav navbar-nav-scroll d-flex me-auto flex-sm-fill"><form class="d-flex" role="search"></form></ul></div></div></nav><div class="container-xxl" role="main"><div class="page-header"><div class="row"><div class="col-sm-11"><h1>Device-Manager </h1></div></div></div><form><nav><div class="nav nav-tabs" id="nav-tab" role="tablist"></div></nav></form></div><div class="tab-pane fade show active" id="nav-pin" role="tabpanel" aria-labelledby="nav-pin-tab" tabindex="0"><div class="card"><div class="card-body"><div class="row mb-3"><label for="inputPinProfile" class="col-sm-2 col-form-label">Selected profile:</label><div class="col-sm-10"></div></div></div></div></div><div class="tab-content" id="nav-tabContent"><div class="tab-pane fade show active" id="nav-pin" role="tabpanel" aria-labelledby="nav-pin-tab" tabindex="0"><div class="card"><div class="card-body"><div class="row mb-3"><div class="col-sm-2"></div><div class="col-sm-10"></div></div><div class="alert alert-danger mt-3" role="alert">Your
 device may stop responding if you select an incompatible profile. In 
this case, you must perform a deletion via the serial interface.</div><div class="card"><div class="card-header text-bg-primary">Connection overview</div><div class="card-body card-text"><div class="table-responsive">
Category | Name | Selected | Active
-- | -- | -- | --
Cmt | clk |   | -1
cs |   | -1
fcs |   | -1
gpio2 |   | -1
gpio3 |   | -1
sdio |   | -1
Display | clk | 22 | 22
cs |   | 255
data | 21 | 21
reset |   | 255
type | 3 | 3
Eth | clk_mode | 0 | 0
enabled | false | false
mdc | -1 | -1
mdio | -1 | -1
phy_addr | -1 | -1
power | -1 | -1
type | 0 | 0
Led | led0 |   | -1
led1 |   | -1
Nrf24 | clk | 18 | 18
cs | 5 | 5
en | 4 | 4
irq | 16 | 16
miso | 19 | 19
mosi | 23 | 23

</div></div></div></div></div></div></div><hr class="border border-3 opacity-75"><div class="d-grid gap-2 d-md-flex justify-content-md-end"></div><div id="app" data-v-app=""></div>
  

</body>
</html>OpenDTU

    [Live View](http://192.168.88.26/)
    [Settings](http://192.168.88.26/settings/device#)
    [Info](http://192.168.88.26/settings/device#)
    [About](http://192.168.88.26/about)
    *

Device-Manager
Selected profile:
Your device may stop responding if you select an incompatible profile. In this case, you must perform a deletion via the serial interface.
Connection overview
Category	Name	Selected	Active
Cmt	clk		-1
cs		-1
fcs		-1
gpio2		-1
gpio3		-1
sdio		-1
Display	clk	22	22
cs		255
data	21	21
reset		255
type	3	3
Eth	clk_mode	0	0
enabled	false	false
mdc	-1	-1
mdio	-1	-1
phy_addr	-1	-1
power	-1	-1
type	0	0
Led	led0		-1
led1		-1
Nrf24	clk	18	18
cs	5	5
en	4	4
irq	16	16
miso	19	19
mosi	23	23

tbnobody, 2024-04-06:
> * Please provide a console log (as text, no screenshot)
> * Please also provide a screenshot of the Device Info Dialog

Sorry, I meant the Inverter info dialog
[image]

yzffrog, 2024-04-07:
hm, sorry i cannot idetify  where i can find the inverter info dialog,  regaring console log: when  i trigger the request and afterwards open the console log - will this show the info you need ?

yzffrog, 2024-04-08:
<html><body>
<h5 class="modal-title">Inverter Info</h5>
Serial | 116183063371
-- | --
Production Year | 2022
Production Week | 30
Model | Unknown model! Please report the "Hardware Part Number" and model (e.g. HM-350) as an issue here.
Detected max. Power | 0 W
Bootloader Version | 0.0.0
Firmware Version | 0.2.56
Firmware Build Date | 1911-09-09 11:52:32
Hardware Part Number | 124097
Hardware Version | 00.00

</body>
</html>Inverter Info
Serial	116183063371
Production Year	2022
Production Week	30
Model	Unknown model! Please report the "Hardware Part Number" and model (e.g. HM-350) as an issue [here](https://github.com/tbnobody/OpenDTU/issues).
Detected max. Power	0 W
Bootloader Version	0.0.0
Firmware Version	0.2.56
Firmware Build Date	1911-09-09 11:52:32
Hardware Part Number	124097
Hardware Version	00.00

looks not as i expected...

yzffrog, 2024-04-09:
Console:
Fetch inverter: 116183063371
08:30:37.808 > TX RealTimeRunData Channel: 3 --> 15 83 06 33 71 80 16 73 88 80 0B 00 66 14 E0 8F 00 00 00 00 00 00 00 00 D2 4C B7 
08:30:38.362 > RX Period End
08:30:38.362 > All missing
08:30:38.362 > Nothing received, resend whole request
08:30:38.362 > TX RealTimeRunData Channel: 23 --> 15 83 06 33 71 80 16 73 88 80 0B 00 66 14 E0 8F 00 00 00 00 00 00 00 00 D2 4C B7 
08:30:38.407 > Interrupt received
08:30:38.461 > RX Channel: 3 --> 95 83 06 33 71 83 06 33 71 01 00 01 01 C3 00 C6 00 CC 03 7C 03 99 00 00 90 A8 80 | -80 dBm
08:30:38.507 > Interrupt received
08:30:38.552 > RX Channel: 61 --> 95 83 06 33 71 83 06 33 71 02 00 00 94 FB 00 53 00 54 01 9F 00 95 00 9E 02 6C 04 | -80 dBm
08:30:38.598 > Interrupt received
08:30:38.650 > RX Channel: 61 --> 95 83 06 33 71 83 06 33 71 03 02 90 00 00 89 1A 00 00 8A 36 00 3D 00 3F 09 47 67 | -80 dBm
08:30:38.894 > RX Period End
08:30:38.894 > Last missing
08:30:38.894 > Request retransmit: 4
08:30:38.894 > TX RequestFrame Channel: 40 --> 15 83 06 33 71 80 16 73 88 84 3B 
08:30:38.944 > Interrupt received
08:30:38.995 > RX Channel: 3 --> 95 83 06 33 71 83 06 33 71 84 13 84 0B 77 00 DC 00 7C 03 E5 00 E5 00 03 3C 7F 19 | -80 dBm
08:30:39.048 > RX Period End
08:30:39.048 > Success
08:30:39.256 > TX AlarmData Channel: 61 --> 15 83 06 33 71 80 16 73 88 80 11 00 66 14 E0 8F 00 00 00 00 00 00 00 00 08 57 6C 
08:30:40.080 > RX Period End
08:30:40.080 > All missing
08:30:40.080 > Nothing received, resend whole request
08:30:40.080 > TX AlarmData Channel: 75 --> 15 83 06 33 71 80 16 73 88 80 11 00 66 14 E0 8F 00 00 00 00 00 00 00 00 08 57 6C 
08:30:40.866 > RX Period End
08:30:40.866 > All missing
08:30:40.866 > Nothing received, resend whole request
08:30:40.866 > TX AlarmData Channel: 3 --> 15 83 06 33 71 80 16 73 88 80 11 00 66 14 E0 8F 00 00 00 00 00 00 00 00 08 57 6C 
08:30:41.664 > RX Period End
08:30:41.664 > All missing
08:30:41.664 > Nothing received, resend whole request
08:30:41.664 > TX AlarmData Channel: 23 --> 15 83 06 33 71 80 16 73 88 80 11 00 66 14 E0 8F 00 00 00 00 00 00 00 00 08 57 6C 
08:30:42.466 > RX Period End
08:30:42.466 > All missing
08:30:42.466 > Nothing received, resend whole request
08:30:42.466 > TX AlarmData Channel: 40 --> 15 83 06 33 71 80 16 73 88 80 11 00 66 14 E0 8F 00 00 00 00 00 00 00 00 08 57 6C 
08:30:42.512 > Interrupt received
08:30:42.559 > RX Channel: 75 --> 95 83 06 33 71 83 06 33 71 81 11 00 00 01 D8 C4 18 | -80 dBm
08:30:43.219 > RX Period End
08:30:43.219 > Success

yzffrog, 2024-04-15:
Anything i can provide in addition ? Any news ?   kr

yzffrog, 2024-05-05:
> :
> @yzffrog <https://github.com/yzffrog>
>
> please click here:
> (follow the red arrow)
> Bildschirmfoto.2024-04-08.um.00.40.17.png (view on web)
>
> greetings,
> metachuh

Thats what i did  its within zje github dialogue

kr greg

yzffrog, 2024-05-24:
[image]

yzffrog, 2024-05-24:
sorry seems was a misundrerstanding, still the same issue

closed 2025-04-18 by github-actions[bot]