tbnobody/OpenDTU #1891: Multiple Time Calibrations on microinverters

Berondal, 2024-04-07:
### What happened?

Hello
since Firmware update from (23.3.xx) i get on two(of 6) microinverters a lot of time calibration.
Microinverter is not reachable for long time, up to 2h.
All 6 inverters:
Modell | HMS-1600-4T
Bootloader-Version | 0.1.0
Firmware-Version | 1.0.18
Firmware-Erstellungsdatum | 2022-08-26 20:51:00
Hardware-Teilenummer | 270680323
Hardware-Version | 01.00

### To Reproduce Bug

no

### Expected Behavior

Look if you can find the reason.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.31

### Relevant log/trace output

```Shell
Begin	Ende	ID	Meldung
06:35:36	06:35:36	1	Wechselrichter gestartet
12:51:06	12:51:06	2	Time calibration
12:51:11	12:51:11	2	Time calibration
12:51:40	12:51:40	2	Time calibration
12:51:46	12:51:46	2	Time calibration
12:52:14	12:52:14	2	Time calibration
12:52:19	12:52:19	2	Time calibration
12:52:48	12:52:48	2	Time calibration
12:52:53	12:52:53	2	Time calibration
12:53:22	12:53:22	2	Time calibration
12:53:27	12:53:27	2	Time calibration
12:54:27	12:54:27	2	Time calibration
12:55:05	12:55:05	2	Time calibration
12:55:36	12:55:36	2	Time calibration
12:55:41	12:55:41	2	Time calibration
```

### Anything else?

Other 4/6 inverters don't give any message "time calibration"

stefledufinfra, 2024-04-09:
Same here with HM400
Firmware : 24.3.31

tbnobody, 2024-04-09:
No one (except Hoymiles) knows when this event occours. Sometimes it occours when when sending commands in a too short time period, sometimes when the RF connection is bad, or when sending invalid packages. It's also not 100% clear if the error text is correct (previously it was defined as "DTU command failed").

stefledufinfra, 2024-04-09:
> No one (except Hoymiles) knows when this event occours. Sometimes it occours when when sending commands in a too short time period, sometimes when the RF connection is bad, or when sending invalid [...]

i never update HOymiles firmware , and i never had this message before i upgrade from 22 to 24 firmware

Peertje-62, 2024-04-11:
Hi, for what it's worth, I get the same 'output' (on both) if I connect an Open-DTU parallel to the Ahoy-DTU .
greeting

tbnobody, 2024-04-11:
> get the same 'output' (on both) if I connect an Open-DTU parallel to the Ahoy-DTU .

Thats normal.... Two DTUs on one inverter lead to crappy data, unknown side effects and other isssues....
And yes, you  see the  same in both projects as the event log is only held in the inverter and both projects read the eventlog directly from the inverter. 
But you already knew that... it's mentioned very big in the documentation: https://www.opendtu.solar

[image]

Peertje-62, 2024-04-11:
Good evening,
You are right I knew this all ready for a long time, but I thought it would or could help to solve the problem of Berodal. since He  has the same output with only one DTU
So apparently not.
THX

tbnobody, 2024-04-11:
Sorry, unfortunately not 😔
It's a strange error code.

puni2k, 2024-04-13:
> No one (except Hoymiles) knows when this event occours. Sometimes it occours when when sending commands in a too short time period, sometimes when the RF connection is bad, or when sending invalid [...]

Sometimes my openDTU device acts up and successively dis- and then reconnects to the MQTT server. I noticed this when looking at the logs of my openHAB server. Simultaneously I also see a similar number of "Time calibration" messages showing up on all my inverters (4).
According to "System Uptime" the ESP itself isn't rebooting.
It doesn't happen often, at average maybe once a month so i just shrug at it.

Berondal, 2024-04-13:
I try to roll back with firmware. But I get some errors. It could be FW 23.3.28

Berondal, 2024-04-14:
I don't know, but I remember, that I have change another thing.
I installt https://github.com/henne49/dbus-opendtu on my Victron Cerbo GX. Could this influence the error?

Berondal, 2024-04-22:
Some days later... Saturday, one inverter don't do anything. wtf,

I tried a lot, ... Today, no missing data, all inverters send data. 
no time calibration

I change CMT2300A Frequenz from 865 MHz to 866 MHz.

Question: If CMT2300A is for connection to HMS/HMT and Hoymiles uses 868 MHz. Why ist it at 865 MHz?
Can I switch to 868 MHz?

markusrudolf, 2024-05-21:
I had the same with any OpenDTU version newer then v24.4.12 with an Hoymiles HM800-2T. Lot's of data loss as well (no response from Inverter for extended period. Inverter is like 2m away from OpenDTU. Power Supply of OpenDTU stabilized with capacitors on RF Module and ESP32 Board.

I Downgraded to 24.4.12, no Issues any longer. I also have no idea what the problem may be. From the changelog it seems like only minor changes to WebGUI and other cosmetics but no changes to communication.

puni2k, 2024-06-10:
Recently I observed an increase of "Time Calibration" messages across my inverters until suddenly my OpenDTU system stopped working entirely.
Got stuck in a boot loop and only came back online after I disconnect Vcc from the NRF board.

Switched to a backup system and for the last few days haven't seen and "Time Calibration" messages. Haven't yet investigated which part was to blame on the old system (NRF board, Capacitor, ESP board). I suspect the latter since that board already gave me headaches when it came to stability on the 3.3V rail.

So I guess that "Time Calibration" is an indicator of intermittent failing communication between DTU and inverter, for whatever reason. In my case likely NRF board dropping out bc of power supply issues (general term, not certainly the brick). ESP32 system uptime looked fine.

closed 2025-04-18 by github-actions[bot]