tbnobody/OpenDTU #1899: Falsche Gesamtleistung

bobbiest, 2024-04-09:
### What happened?

Schaltet bei zwei installierten Wechselrichtern einer ab (beispielsweise wegen Verschattung oder Ausrichtung der Module), wird der letzte produzierte Wert weiterhin zur Gesamtleistung addiert.
[image]
[image]

### To Reproduce Bug

Zwei Wechselrichter, bei einem die Module verschatten, der andere produziert weiterhin.  

### Expected Behavior

Wenn ein Wechselrichter abgeschaltet ist bzw keinen Ertrag mehr liefert, sollte nur der Wert vom aktiven Wechselrichter angezeigt werden.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.31

### Anything else?

Auch in früheren Versionen wurde dies bereits beobachtet.

tbnobody, 2024-04-09:
Aktuell werden alle Inverter aufsummiert die in den Invertereigenschaften aktiviert sind (poll enabled)

https://github.com/tbnobody/OpenDTU/blob/12588655dfa9b06aff4bff04ee1d1a2280dda525/src/Datastore.cpp#L94-L99

Aber du hast vmtl. recht. Ggf. müsste hier noch abgefragt werden ob der entsprechende Inverter erreichbar ist. Bin gerade nicht sicher ob es einen Use-Case gibt diesen Wert trotzdem aufzusummieren.

Alternativst kannst du auch die Einstellung "Zero Runtime Data" aktivieren. Dann werden Laufzeit Daten des Inverters auf 0 gesetzt wenn dieser nicht erreichbar ist.

bobbiest, 2024-04-09:
Danke für die schnelle Rückmeldung und generell für die mega Arbeit!

Zero Runtime Data reicht mir als Workaround, da tagsüber bei ausreichend Sonne ein Akku geladen wird, statt einzuspeisen.

Beste Grüße

closed 2024-04-10 by tbnobody