tbnobody/OpenDTU #1902: Restarting the inverter via home assistant and mqtt does not work.

Lice2, 2024-04-10:
### What happened?

I integrated OpenDTU into Home Assistant via MQTT. The values etc are also all displayed. It only doesn't work if I want to turn the inverter on, off or restart via Home Assistant.
This works without any problems via OpenDTU.

### To Reproduce Bug

[image]
Wenn ich hier an den Reglern was ändere oder auf den Wechselrichter Ein / Aus oder neustarten klicke passiert nichts.
118 / 5.000
Übersetzungsergebnisse
Übersetzung
If I change something on the controls here or click on the inverter on/off or restart, nothing happens

[image]

### Expected Behavior

My MQTT-Broker-Parameter in OpenDTU:

MQTT aktivieren : On
Home Assistant MQTT-Auto-Discovery aktivieren: On

Hostname: : Ip from HA
Port: 1883
Username: xxxxxxx
Password: xxxxx
Basis-Topic: einspeisung/
Veröffentlichungsintervall: 5
CleanSession Flag aktivieren: On
Retain Flag aktivieren: On
TLS aktivieren: Off

LWT-Parameter
LWT-Topic: einspeisung/  dtu/status
LWT-Online-Nachricht: online
LWT-Offline-Nachricht: offline
QoS (Quality of Service): 2(exakt einmal)

Home Assistant MQTT-Auto-Discovery-Parameter
Präfix Topic: homeassistant/
Retain Flag aktivieren : on
Ablauffunktion aktivieren: on
Einzelne Panels: on

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.31

tbnobody, 2024-04-10:
* If you look at your broker with mqtt explorer or mosquitto_sub, do you see a change in the `einspeisung/[serial]/cmd/power` topic?
* If you look at the opendtu console while you press restart, do you see anything regarding a received Mqtt message?

Lice2, 2024-04-10:
in opendtu in the console:

23:18:04.920 > Success
23:18:09.104 > Fetch inverter: 114184804642
23:18:09.218 > TX RealTimeRunData Channel: 3 --> 15 84 80 46 42 80 14 73 60 80 0B 00 66 17 02 10 00 00 00 00 00 00 00 00 FC 0B 8D 
23:18:09.269 > Interrupt received
23:18:09.270 > RX Channel: 40 --> 95 84 80 46 42 84 80 46 42 01 00 01 00 07 00 02 00 00 00 F5 00 06 00 0F 00 00 6C | -80 dBm
23:18:09.284 > Interrupt received
23:18:09.297 > RX Channel: 75 --> 95 84 80 46 42 84 80 46 42 02 00 00 00 00 00 0F 00 00 00 03 00 08 00 00 00 00 93 | -80 dBm
23:18:09.484 > RX Period End
23:18:09.484 > Last missing
23:18:09.484 > Request retransmit: 3
23:18:09.484 > TX RequestFrame Channel: 23 --> 15 84 80 46 42 80 14 73 60 83 11 
23:18:09.528 > Interrupt received
23:18:09.541 > RX Channel: 61 --> 95 84 80 46 42 84 80 46 42 83 00 00 00 00 00 00 00 A5 00 05 5C A5 4F | -80 dBm
23:18:09.584 > RX Period End
23:18:09.584 > Success
23:18:13.957 > Fetch inverter: 114184804642
23:18:13.980 > TX RealTimeRunData Channel: 40 --> 15 84 80 46 42 80 14 73 60 80 0B 00 66 17 02 15 00 00 00 00 00 00 00 00 AC 34 E7 
23:18:14.017 > Interrupt received
23:18:14.049 > RX Channel: 75 --> 95 84 80 46 42 84 80 46 42 01 00 01 00 07 00 02 00 00 00 F4 00 06 00 0F 00 00 6D | -80 dBm
23:18:14.500 > RX Period End
23:18:14.500 > Last missing
23:18:14.500 > Request retransmit: 2
23:18:14.500 > TX RequestFrame Channel: 61 --> 15 84 80 46 42 80 14 73 60 82 10 
23:18:14.606 > RX Period End
23:18:14.606 > Last missing
23:18:14.606 > Request retransmit: 2
23:18:14.606 > TX RequestFrame Channel: 75 --> 15 84 80 46 42 80 14 73 60 82 10 
23:18:14.714 > RX Period End
23:18:14.714 > Last missing
23:18:14.714 > Request retransmit: 2
23:18:14.714 > TX RequestFrame Channel: 3 --> 15 84 80 46 42 80 14 73 60 82 10 
23:18:14.823 > RX Period End
23:18:14.823 > Last missing
23:18:14.823 > Request retransmit: 2
23:18:14.823 > TX RequestFrame Channel: 23 --> 15 84 80 46 42 80 14 73 60 82 10 
23:18:14.867 > Interrupt received
23:18:14.883 > RX Channel: 61 --> 95 84 80 46 42 84 80 46 42 02 00 00 00 00 00 0F 00 00 00 03 00 07 00 00 00 00 9C | -80 dBm
23:18:14.936 > RX Period End
23:18:14.936 > Last missing
23:18:14.936 > Request retransmit: 3
23:18:14.936 > TX RequestFrame Channel: 40 --> 15 84 80 46 42 80 14 73 60 83 11 
23:18:14.965 > Interrupt received
23:18:14.982 > RX Channel: 23 --> 95 84 80 46 42 84 80 46 42 83 00 00 00 00 00 00 00 A5 00 05 98 87 A9 | -80 dBm
23:18:15.028 > RX Period End
23:18:15.028 > Success

Lice2, 2024-04-10:
in mqtt-mosquitto 
2024-04-10 22:55:35: New connection from 192.168.0.192:50100 on port 1883.
2024-04-10 22:55:35: New client connected from 192.168.0.192:50100 as OpenDTU-16247360 (p2, c1, k15, u'xxxx').

tbnobody, 2024-04-10:
If this is the only output while you are pressing restart in home assistant, home assistant is not allowed to publish to the broker. Otherwise mosquitto_sub would show the output. Therefor you should search the issue between home assistant and mosquitto (and if this works look further at opendtu)

Lice2, 2024-04-10:
OK,
I haven't found anything yet...

tbnobody, 2024-04-11:
Have you configured any type of ACL's? 
I don't know your setup. Have you a Home Assistant Core installation and installed mosquitto manually or are you using HAOS and using the mosquitto as Addin?

Lice2, 2024-04-11:
I used HAOS and installed the addin.
What are acl's?

Lice2, 2024-04-12:
I have now installed MQTT Explorer under HOmeAssistant.
If I click the buttons now it says.
CMD
Power = 1 or Power = 0.
When I press the buttons to turn on or off
[image]

tbnobody, 2024-04-12:
> What are acl's? ACL = Acces Control List ?

Yes

> I only have the data from the mosquitto broker that I entered in openDTU.

That means you are not the administrator of the broker? In either way you have to make sure, that the user `liceqtt` has `readwrite` permissions to the topic `einspeisung/#` (which should be the default configuration if you didn't change anything)

* Have you tried restarting OpenDTU (just in case)
* When looking at the console output above, did you capture it at the exact same time as you pressed the power or restart button? When the publish is received by OpenDTU you should immediately see a output on the console
* Did you do any special configuration at broker side or OpenDTU MQTT side or is everything at default values?

Lice2, 2024-04-12:
[image]

OK, now it works.
I gave liceqtt admin.
Now I can also restart OpenDTU, change values, etc.
Many thanks for the support !!!!

closed 2024-04-12 by Lice2