tbnobody/OpenDTU #1911: Entity for string voltage

IenteredanUsername, 2024-04-12:
### Is your feature request related to a problem? Please describe.

It would be great to obtain entities for the voltage of a string, allowing me to determine the state of charge based on the voltage of a connected battery to the inverter and display it in Home Assistant. Thank you so far!

### Describe the solution you'd like

Add entity for connected strings

tbnobody, 2024-04-12:
Just enable the individual panels option...
https://www.opendtu.solar/firmware/configuration/mqtt_settings/#individual-panels

closed 2024-04-12 by tbnobody

IenteredanUsername, 2024-04-13:
perfekt, thank you !