tbnobody/OpenDTU #1919: Fusion OpenDTU does not connect to HMS-1600-4T

floriangutzeit, 2024-04-18:
### What happened?

Occasionally, after external events (e.g. grid loss, firmware update, converter limit changed overnight, ...) the Fusion OpenDTU from 3PrintD Solution does not find the connection to anymore HMS-1600-4T during the morning. After some hours, even the LAN connection to the OpenDTU gets lost, so I have to pull the plug in order to restart it. It takes weeks until the connection is suddenly back again. The converter is producing, flash light is blinking green. I have ensured the frequency (865 MHz) has sufficient distance to my neighbor with the same converter type, firmware has the latest updates... What else can I do?

### To Reproduce Bug

So far, this happened after e.g. a grid loss, firmware update, converter limit changed overnight. I cannot really reproduce it on purpose, neither I can reproduced why it is suddenly working again after weeks.

### Expected Behavior

After above mentioned events, OpenDTU goes back to work.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.4.12

### Relevant log/trace output

```Shell
RX Period End
08:23:01.513 > All missing
08:23:01.513 > Nothing received, resend whole request
08:23:01.513 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 01 04 00 00 5D 41 CA 
08:23:03.562 > RX Period End
08:23:03.562 > All missing
08:23:03.562 > Nothing received, resend count exeeded
08:23:03.565 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 BE 00 00 44 61 48 
08:23:05.612 > RX Period End
08:23:05.612 > All missing
08:23:05.612 > Nothing received, resend whole request
08:23:05.612 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 BE 00 00 44 61 48 
08:23:07.623 > RX Period End
08:23:07.623 > All missing
08:23:07.623 > Nothing received, resend whole request
08:23:07.623 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 BE 00 00 44 61 48 
08:23:09.708 > RX Period End
08:23:09.708 > All missing
08:23:09.708 > Nothing received, resend whole request
08:23:09.708 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 BE 00 00 44 61 48 
08:23:11.756 > RX Period End
08:23:11.756 > All missing
08:23:11.756 > Nothing received, resend whole request
08:23:11.756 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 BE 00 00 44 61 48 
08:23:13.701 > RX Period End
08:23:13.702 > All missing
08:23:13.702 > Nothing received, resend count exeeded
08:23:13.707 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 E6 00 00 97 E0 42 
08:23:15.851 > RX Period End
08:23:15.851 > All missing
08:23:15.851 > Nothing received, resend whole request
08:23:15.851 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 E6 00 00 97 E0 42 
08:23:17.899 > RX Period End
08:23:17.899 > All missing
08:23:17.899 > Nothing received, resend whole request
08:23:17.899 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 E6 00 00 97 E0 42 
08:23:19.845 > RX Period End
08:23:19.845 > All missing
08:23:19.845 > Nothing received, resend whole request
08:23:19.845 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 E6 00 00 97 E0 42 
08:23:21.892 > RX Period End
08:23:21.892 > All missing
08:23:21.892 > Nothing received, resend whole request
08:23:21.892 > TX ActivePowerControl 865.00 MHz --> 51 91 62 67 06 80 13 01 08 81 0B 00 00 E6 00 00 97 E0 42
```

### Anything else?

[image]
[image]
[image]
[image]

mager33, 2024-04-27:
this version seems unstable, see other bug report; update to 24.4.24!

tbnobody, 2024-05-04:
Duplicate of #1921

closed 2024-05-04 by tbnobody