tbnobody/OpenDTU #1927: Avoid accidental downgrade (=bricking the device)

broth-itk, 2024-04-20:
### Is your feature request related to a problem? Please describe.

Well, it happed to me: I wanted to upgrade to the latest release but somehow I selected accidently an older version from the downloads folder.
The device rebooted and was obviously not available anymore.

I had to perform a complete reflash with the factory image and restore the configuration from backup.

### Describe the solution you'd like

This problem might be easy to fix but some OpenDTU units might be installed in a not-so-easy-to-reach location.

I would like to suggest to

- Include some kind of version number (integer) in an fixed address of the image, best somewhere at the beginning
- Running OpenDTU checking if own version > upload version, then discard
- If upload version is newer, proceed with the update as usual

Lets discuss the impact or maybe other ideas which would help.
Thanks!

tbnobody, 2024-04-20:
I don't like the idea as it would prevent all users to downgrade if a version contains a mistake.

broth-itk, 2024-04-21:
I'm not advocating against the possibility to downgrade. Don't get me wrong.
I just want to avoid the device getting bricked for some reason due to a mishap.

@MetaChuh The unit was running v24.3.31 and instead of installing v24.4.12 I uploaded v23.12.18

broth-itk, 2024-04-21:
What about a new version field which indicates the "generation" (including memory layout, configuration file format and so on):
- Upgrades and downgrades within the same generation is possible.
- Upgrade to newer generation only
- No downgrade to older generation

tbnobody, 2024-04-21:
The downgrad to this version failed because of a bug in the upstream esp32 framework. 
Feel free to create a PR implementing this checks. But then you also have to check every release and check to which previous release it's possible to downgrade.
I don't have the resources to do this.

broth-itk, 2024-04-21:
This is not meant to be a "Arbeitsbeschaffungsmaßnahme" (work creation plan).

If there was an efficient way for the software to detect an older version, a message like "warning: attempting downgrade, are you sure" is sufficient.
... or a checkbox "allow downgrades" in the Firmware Upgrade section.

Any other ideas?