tbnobody/OpenDTU #1931: [Request] Add support to TSOL-MX*

liberodark, 2024-04-21:
### Is your feature request related to a problem? Please describe.

Hi,

Can you add support to TSOL-MX400 TSOL-MX450 TSOL-MX500.

Best Regards

### Describe the solution you'd like

I'm going to purchase play 2 with a SOL-MX450.
So I could perhaps help with the tests also I have an openDTU.

Franckled1, 2024-10-30:
Hello,
I am in the same configuration than liberodark.
Can you add TSOL MX450 to current inverters ?
Have a good day

stefan123t, 2024-10-30:
According to the datasheet that @liberodark linked the TSUN TSOL-MX series models have a Communication Module: WiFi.

I assume the Gen3 here does not mean it is using Hoymiles Gen3 RF Protocol but rather that it is similar to the HMS-WiFi series models with an integrated DTU-BI (built-in).

You may refer to the respective projects for mimicking an S-Miles Installer App / S-Miles Cloud from @DennisOSRM and @rovo89 in their own repositories.