tbnobody/OpenDTU #1936: Bei einphasigen Wechselrichtern ist Angabe "Phase 1" irreführend

Gitsarry, 2024-04-22:
### Is your feature request related to a problem? Please describe.

Mir geht es genauso wie dem Poster von #1044 welches leider vom Bot geschlossen wurde.

Ich habe zwei HM-Wechselrichter, die jeweils einphasig angeschlossen sind.

Nun fragte ich mich, ob die beiden WR an unterschiedlichen Phasen angeschlossen wären, was mich besser dünkte.

Und erinnerte mich, dass die OpenDTU anzeigte, an welcher Phase ein WR angeschlossen sein. Enttäuscht stellte ich fest, dass die OpenDTU beide WR als "Phase 1" anzeigte, und missverstand diese Anzeige dahingehend, dass beide WR an derselben Phase angeschlossen seien. Ich suchte dann aber nochmal nach dem Thema im Internet und stieß dann auf #1044 wodurch ich lernte, so habe ich es jetzt verstanden, dass "Phase 1" meint, dass der WR an genau einer Phase hängt. Was bei einem einphasigen WR eine überflüssige Information ist. Ich stelle mir vor dass bei einem dreiphasigen WR dann man dort die unterschiedlichen Module angeschlossen sieht an Phase 1, Phase 2 und Phase 3, OK.

Aber mich leitete die Angabe "Phase 1" in meinem Fall wie beschrieben in die Irre.

### Describe the solution you'd like

Bei einphasigen WR anstatt "Phase 1" angeben "einphasiger Anschluss".

### Describe alternatives you've considered

Bei einphasigen WR Angabe "einphasig" oder Angabe zu Phase ganz weglassen.

### Additional context

Mit der Messmethode [Wie kriege ich raus, ob zwei Steckdosen an derselben Phase hängen?](https://www.mikrocontroller.net/topic/294794#3143659) konnte ich feststellen, dass meine beiden WR an unterschiedlichen Phasen hängen, obwohl für beide in OpenDTU "Phase 1" angezeigt wird.

Charly-2, 2024-04-22:
Die Verwirrung gibts auch bei 3phasigen HMT WR. Dort gibt es auch nur eine "Phase 1" . Siehe Diskussionen.
"AC" wie bei MQTT würde beide Situationen korrekt beschreiben.
Wenn bei HMT dann irgendwann 3 Phasen reported werden könnte man "AC Phase 1" usw. verwenden.
PS: Danke für diese super Software!
Charly

z-master42, 2024-04-24:
Woher soll die DTU wissen an welcher Phase der Wechselrichter hängt?

Gitsarry, 2024-04-24:
> Woher soll die DTU wissen an welcher Phase der Wechselrichter hängt?

Die Frage ist off topic. Wenn die DTU jedenfalls nicht weiß an welcher Phase ein Wechselrichter hängt, sollte für den Wechselrichter nicht "Phase 1" angezeigt werden, denn das hinterlässt beim User den Eindruck, der Wechselrichter würde an einer bestimmten Phase hängen (z.B. es würde eine "Phase 2" geben, an der der Wechselrichter _nicht_ hängen würde, dies ist jedoch nicht der Falls wie ich beschrieben habe).

turrican944, 2024-04-26:
Da steht Phase 1 das finde ich aus technischer Sicht auch völlig korrekt, das sagt halt aus der WR nur eine AC Phase hat. Gut der HMT sollte auch 2 und 3 mit anzeigen. Damit weis man (und der WR) immer noch nicht welche Phase wirklich 1 ist. Wie soll er das auch rausbekommen. Letztendlich ist L1 L2 L3 auch nur eine definition. Ob deine L1 im Haus wirklich an L1 im Trafo hängt weist du auch nicht. Das Drehfeld ist idr. rechts drehend.
Natürlich kann man auf einem 1 Phasigen WR auch einfach nur AC Spannung oder so anzeigen und Phase 1 da weglassen aber ich stelle mir die Frage warum, und auf die Idee des Themenstarter das Phase 1 sich auf die angeschlossen Phase bezieht bin ich im leben nicht gekommen (liegt vielleicht auch daran das ich als Elektrotechniker auf solche ideen nicht kommen weil mir klar ist das WR das nicht ermitteln kann).

SolarFreund, 2024-10-28:
Hallo,

Frage eine Newcomers:

ein Noah2000  mit eingestellten 100 Watt speist einen HMS-2000.

abgegeben DC-Leistung mit 98,xx Watt korrekt  - ist die Summe der Werte von  String1+String2
**bei Phase1  werden nur 46,7 Watt angezeigt**   

Das passt irgenwie zu nix... ?
Übersehe ich was ?

(Habe ein openDTU-Fertiggerät lauffähig im WWW gekauft, kann da was falsch geflasht sein? )

Danke für jeden Hinweis.

stefan123t, 2024-10-28:
@turrican944 @Gitsarry bei den Original Hoymiles DTUs kann man für jeden WR angeben auf welcher Phase dieser WR seinen Saft einspeist. Das ist vor allem im Hinblick auf eine Dynamic Power Limit Funktion relevant, da damit die Verbraucher / der Verbrauch auf einer Phase besser direkt mit Hilfe der an diese Phase angeschlossenen WR kompensiert werden kann. Dazu gibt es in der Hoymiles Config ein Flag Phaee A/B/C.
Es ist mE nicht gabz abwegig diese Information auch in den Inverter Einstellungen zu hinterlegen und die Summe für Phase 1,2,3 oder A/B/C dann auf der Live View Seite bzw in den Daten für die REST API bzw MQTT entsprechend der Phase aufzusummieren.
M.E. Ein Feature Request für ein Flag Phase bei jedem einp-hasigen WR und bei einem drei-phasigen sollte man es evtl. auch je nach Anschluss zuordnen können ?

z-master42, 2024-10-28:
> Hallo,
>
>
> Frage eine Newcomers:
>
> ein Noah2000  mit eingestellten 100 Watt speist einen HMS-2000.
>
> abgegeben DC-Leistung mit 98,xx Watt korrekt  - ist die Summe der Werte von  String1+String2
> **bei [...]

@SolarFreund Das hat nichts mit der DTU zu tun.
Welchen Wirkungsgrad erreicht der HMS denn? Bestimmt keine 95 %, wohl er so 47 %.
Musst mit der DC-Leistung höher gehen, damit der Wirkungsgrad steigt, dann kannst du dich wieder nach unten tasten und gucken, ob er stabil bleibt.

stefan123t, 2024-10-29:
Hier ist der entsprechende Dialog zur Zuordnung der Phase A/B/C aus der S-Miles Cloud:

[image]

SolarFreund, 2024-10-31:
Danke für die Hilfen.

Habe die Strings benannt und dann ist **nach Neustart der openDTU  die Anzeige  richtig**.

1. Schaltung nicht angefasst
2. in der Live-Ansicht sind **nun** aber auch die nicht belegten Strings 3 und 4 nicht mehr da

Also für mich alles ok !