tbnobody/OpenDTU #1948: Wert bleibt beim ausgeschalteten Wechselrichter stehen.

Arkadier, 2024-04-27:
### What happened?

Nachdem der Wechselrichter aus ist, bleibt (vermutlich der letzte übermittelte) Wert stehen. Bild 1
Erst nach einem Neustart der DTU ist der Wert bei 0. Bild 2
[image]
[image]

### To Reproduce Bug

Jeden Abend.

### Expected Behavior

Sollte ohne Neustart 0 Watt anzeigen, wenn der Wechselrichter aus ist.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.4.12

tbnobody, 2024-04-27:
Das ist kein bug, das ist gewollt so. Aber schaue dir bitte #1923 an.

Arkadier, 2024-04-28:
In dem Fall sorry, mein Fehler. Das war so, dass bloß ein DTU sich so verhalten hat, eine andere nicht. Die Einstellungen bei beiden DTUs waren die gleichen. Die Nullung war bei beiden ausgeschaltet.

closed 2024-04-28 by Arkadier

Arkadier, 2024-05-01:
Das war beides 24.4.12.
Der einzige Unterschied ist, dass eine mit dem Chip ESP32-D0WD-V3 ist und die, welche die "Probleme" breitet hat den Chip ESP32-D0WDQ6 hat.