tbnobody/OpenDTU #1951: AsyncTCP parameter name change

hawa-lc4, 2024-04-28:
### What happened?

patch file "event_queue_size.patch" included in OpenDTU package fails to install patch with error:
     error: corrupt patch at line 5

### To Reproduce Bug

- perform a full clean to remove  all libraries currently downloaded.
- next download libraries newly
- see Library Manager installing current version 3.1.2 of AsyncTCP:
        Library Manager: Installing mathieucarbou/Async TCP @ ^3.0.2
        Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80%
        Library Manager: Async TCP@3.1.2 has been installed!
- any operation triggering the extra script   pre:pio-scripts/patch_apply.py   reports an error message about the patch script to be failed

### Expected Behavior

Expected behavior is no error message on such operation described above.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

g5b5a984

### Relevant log/trace output

```Shell
Firmware Revision: v24.4.24-1-g5b5a984
error: corrupt patch at line 5
Working on patch: G:\github_repo\OpenDTU\patches\async_tcp\event_queue_size.patch... failed
```

### Anything else?

The new version 3.1.2 of AsyncTCP already has such patch installed, however the defined parameter is named slightly different.
OpenDTU uses   "CONFIG_ASYNC_TCP_EVENT_QUEUE_SIZE"  vs.   "CONFIG_ASYNC_TCP_QUEUE_SIZE"   used in AsyncTCP library.
It appears the event_queue_size.patch is no longer needed and the parameter name should be changed in platformio.ini file accordingly.

tbnobody, 2024-04-28:
Cannot reproduce... I cleaned all files in `libdeps` folder and let all download again during build process

``` 
Working on patch: D:\Projects\Solaranlage\software\OpenDTU\patches\async_tcp\event_queue_size.patch... applied
```

And please keep in mind, the patch will NOT be applied to the `Async TCP` library. It will applied to `AsyncTCP-esphome`.

But you are right, the new define `CONFIG_ASYNC_TCP_QUEUE_SIZE` must also be used.

schneeer, 2024-04-29:
Hi,
I also had a similar error in line 5 when pulling/fetching the last version (4/24/24), but I can't remember it 100%. I also can't tell if it was the same file.

Line 5 was apparently empty.
I put a space, deleted it again, saved the file, then the error message went away and my forked repository was updated.

hawa-lc4, 2024-04-29:
@tbnobody thanks for your quick response.
Cleaning out all files from `libdebs` folder was also the first thing I did but didn't help for me.
Now I've removed my complete repository and perfromed a fresh clone of your `master` repository and the issue remains all the same.

You're right, the patch is about `AsyncTCP-esphome` and not about `Async TCP` as I assumed incorrectly in my bug report.
I'm running out of ideas what's going wrong with the new version of OpenDTU for I had already a successful compilation of the previous release.

@schneeer you're right it seems to be about white spaces in the patch file.
When adding one white space at line 5 the next corrupted line is 19 an d at last 25.
Placing a white space at the beginning of all three lines the patch fails fully with:
`Firmware Revision: v24.4.24
error: patch failed: .pio/libdeps/generic_esp32/AsyncTCP-esphome/src/AsyncTCP.cpp:97
error: .pio/libdeps/generic_esp32/AsyncTCP-esphome/src/AsyncTCP.cpp: patch does not apply
error: patch failed: .pio/libdeps/generic_esp32/AsyncTCP-esphome/src/AsyncTCP.h:53
error: .pio/libdeps/generic_esp32/AsyncTCP-esphome/src/AsyncTCP.h: patch does not apply
Working on patch: G:\github_repo\OpenDTU\patches\async_tcp\event_queue_size.patch... failed`

My last choice is applying the patch manually, removing the patch file and make sure not cleaning `libdebs` folder again after this.
Or are there any other ideas?

schneeer, 2024-04-29:
> Sie haben Recht, es scheint sich um ein Leerzeichen in der Patch-Datei zu handeln.
> Wenn in Zeile 5 ein Leerzeichen hinzugefügt wird, ist die nächste beschädigte Zeile 19 und zuletzt 25. Wenn
> am Anfang aller drei Zeilen ein Leerzeichen eingefügt wird, schlägt der Patch vollständig fehl mit:

Ja genau, so oder ähnlich war das bei mir auch, Ich habe nur leider wenig Ahnung und war froh, das ich das irgendwie hinbekommen habe.
Wie gesagt, dummerweise weis ich nicht mehr genau was ich alles angestellt habe, irgendwann klappte  das.
Somit kann ich nicht sagen, wie ich es tatsächlich behoben habe.

tbnobody, 2024-11-07:
This should be already fixed as the latest versions are using a different fork from async tcp and async webserver.