tbnobody/OpenDTU #1952: Firmware update from v24.3.15 to any higher version via web interface says sucessful but version doesnt increase in system info

Rainking80, 2024-04-28:
### What happened?

i cannot update v24.3.15 to any higher version via web interface. It finishes sucessfully but version doesnt increase in system info. i tried several versions.

### To Reproduce Bug

run a firmware update

### Expected Behavior

updating to v24.4.24

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.3.15

Rainking80, 2024-04-28:
[image]
[image]

tbnobody, 2024-11-07:
Going to close this issue due to inactivity. 
Regarding the connection reset. I also face this issue sometimes. But its still unknown when the issue occours and how to fix it.

closed 2024-11-07 by tbnobody