tbnobody/OpenDTU #1958: Problem beim ändern des Abfrage Intervalls

alfprae, 2024-04-30:
### What happened?

Beim ändern des Abfrage Intervalls kommt folgende Meldung: Die Frequenz muss zwischen 860250000 und 923500000 kHz liegen und ein Vielfaches von 250kHz betragen!
Neuladen der Webseite bringt nichts, bereits auf 3 verschiedenen Geräten versucht.

### To Reproduce Bug

Neues Interval eingeben und speichern. Es gibt auch kein Feld mit der Frequenz.

### Expected Behavior

Ess kommt die Fehlermeldung Die Frequenz muss zwischen 860250000 und 923500000 kHz liegen und ein Vielfaches von 250kHz betragen! beim speichern

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.4.24

tbnobody, 2024-04-30:
What RF module are you using? I assume it's a CMT2300a?

Could you please provide some screenshots of "Info" --> "System" and the "Settings" --> "DTU" view?

alfprae, 2024-04-30:
Rf24

alfprae, 2024-04-30:
[image]
Es ist ein nRF24 Modul

alfprae, 2024-04-30:
[image]
[image]

alfprae, 2024-04-30:
Auf allen 3 Geräten, Win11, Win10 und Android jeweils den Cache geleert und die Seite neu geladen. Hat nichts gebracht.

tbnobody, 2024-04-30:
Hmmm kann ich leider immer noch nicht nachvollziehen. Klappt hier ohne Probleme. Mir ist noch nicht ganz klar was da passiert.

Könntest du ggf. über die Konfigurationsverwaltung ein Backup der config.json erstellen, mit einem Texteditor bearbeiten, da drin dein WiFi Passwort bzw. SSID löschen und die Config hier einstellen?

tbnobody, 2024-04-30:
Also der Fehler befindet sich schon in der config.json. 
Speziell:
```json
"cmt_frequency": 1711573504,
```
Ich bin gerade nicht sicher wie das passiert ist. War das eine Neuinstallation von OpenDTU oder wurde upgegraded?

Als Workaround könntest du versuchen, den Wert in der json Datei (aber Vorsicht, die korrekte Datei ohne herausgelöschte Passwörter etc.) durch folgendes zu ersetzen:
```json
"cmt_frequency": 865000000,
```

alfprae, 2024-04-30:
Es war eine alte Version drauf, diese habe ich geflasht auf die v24.1.26 und später ein Update auf die aktuelle Version gemacht. Der Fehler ist auch schon bei der v24.1.26 aufgetreten. Ich habe deien Workaround durchgeführt. Der Fehler ist weg, ich kann die Änderungen für die Einstellungen der DTU wieder speichern.
[image]

Vielen Dank für die Lösung.

alfprae, 2024-04-30:
Nochmalls vielen Dank

closed 2024-04-30 by alfprae