tbnobody/OpenDTU #1959: Zufälliges zurücksetzten des "Gesamtertrag Heute"

Maschler, 2024-05-03:
### What happened?

Hallo zusammen,
erstmal: Vielen Dank, dass es so ein tolles Projekt gibt!!!

Mir ist in den letzten Tagen aufgefallen, dass dich der Wert "Gesamtertrag Heute" zufällig zurücksetzt. Aufgefallen ist mir das in der grafischen Darstellung.

Beispiel von heute:
Aktueller Wert (16:30)
[image]

Der geloggte Verlauf sieht so aus:
[image]

Wenn ich mir bis 12:30 die Werte anschaue, dann ist der Wert "Gesamtertrag Heute" größer als es im obersten Bild die DTU anzeigt.
[image]

Das Zurücksetzen muss irgendwann zwischen 12:30 und 13:00 passiert sein, weil der Wert plötzlich unter 130 Wh liegt.

### To Reproduce Bug

Das weiß ich leider nicht. Teilweise tritt dies mitten am Tag auf, teilweise nachmittags.

### Expected Behavior

Kein zurücksetzten des "Gesamtertrag Heute"

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.4.24

tbnobody, 2024-05-04:
Hast du die "Nulle Laufzeit Daten" Funktion aktiv? Dann versuch mal den "Erreichbarkeit Schwellenwert" höher zu setzen.

Maschler, 2024-05-04:
Hi,

also die "Nulle Laufzeit Daten" Funktion ist aus. Der "Erreichbarkeit Schwellenwert"  war auf zwei. Da es heute wieder aufgetreten ist (diesmal nach 14:00) habe ich mir die Daten angesehen. 

Dabei habe ich gesehen, dass fast 2 Minuten keine Werte angekommen sind. 
[image]

Im Log war nichts zu sehen. Darum gehe davon aus, dass da Probleme in der Kommunikation (Funkstörung wie auch immer) vorliegen und deshalb die Daten nicht angekommen/abgefragt werden können.

Ich würde mal den "Erreichbarkeit Schwellenwert" hoch setzen. Wie lange ist den die Zeitdauer zwischen 2 Anfragen?

tbnobody, 2024-05-04:
> Wie lange ist den die Zeitdauer zwischen 2 Anfragen?

Das kannst du selbst festlegen: https://www.opendtu.solar/firmware/configuration/dtu_settings/#poll-interval

tbnobody, 2024-05-04:
Aber äh, wenn der Gesamtertrag zurückgesetzt wird, bedeutet das eher das in dem Moment der Inverter neu startet. Dieser Wert kommt nur und ausschließlich direkt aus dem Inverter. 
Hier kannst du aber die Tagesertragskorrektur aktivieren: https://www.opendtu.solar/firmware/configuration/inverter_settings/#yield-day-correction

mroenne2022, 2024-05-05:
Ich hatte heute genau dieses Problem auch ,vorher noch nie vorgekommen. Ebenfalls Version vom 24.4. ! 
Hab an den Erreichbarkeitsdaten keine Einstellungen vorgenommen. Bei beiden WR ist der Tagesertrag auf 0 gegangen,  ich glaube nicht das beide WR neu gestartet sind ,das wäre schon ein großer Zufall. Stromausfall war auch nicht. Hab auch nichts besonderes im Einsatz,  lese lediglich die Erträge übers Webinterface per Handy oder Laptop aus. Vielleicht liegt es doch an der Version !?

tbnobody, 2024-05-05:
In dem Moment in dem im Webfrontend 0 angezeigt wird bitte auf die Konsole schauen und einen längeren Ausschnitt bereitstellen.

Wie werden die Daten abgerufen? Via MQTT?

mroenne2022, 2024-05-05:
Nein , nicht MQTT , das ist garnicht aktiv. 
Ich lese die Daten lediglich manuell per Browser auf die IP und WebUI aus. Insofern hab ich den Zustand heute zum ersten Mal zufällig gesehen und mich and dieses Issue erinnert. 
Im Nachgang kann ich doch jetzt nichts mehr auf der Console dazu sehen oder doch ?

Maschler, 2024-05-06:
> Aber äh, wenn der Gesamtertrag zurückgesetzt wird, bedeutet das eher das in dem Moment der Inverter neu startet. Dieser Wert kommt nur und ausschließlich direkt aus dem Inverter. Hier kannst du aber [...]

Also bei mir stand in den Logs nichts von einem Neustart.
Ich denke, eine alternative Möglichkeit ist, dass die Kommunikation zwischen WR und ESP aus irgendeinem Grund kurzzeitig nicht funktioniert, sodass die "Erreichbarkeit Schwellenwert" gerissen wird.

> In dem Moment in dem im Webfrontend 0 angezeigt wird bitte auf die Konsole schauen und einen längeren Ausschnitt bereitstellen.
> 
> Wie werden die Daten abgerufen? Via MQTT?

Aus meiner Sicht hat das nichts mit MQTT zu tun. Einerseits ist das Qos 2, weshalb die Daten auf jeden Fall ankommen. Andererseits ist MQTT ja nur eine Weitergabe der Daten, weshalb selbst bei nicht Erreichen des Brokers der "Gesamtertrag Heute" nicht auf 0 gesetzt werden dar.

> Ich hatte heute genau dieses Problem auch ,vorher noch nie vorgekommen. Ebenfalls Version vom 24.4. ! 

Genauso bei mir. Das ist erst ab der Version v24.4.24 aufgetreten.

mroenne2022, 2024-05-06:
@Maschler : Ich glaube eher in dieser Version ist irgendein Problem.  Meine beiden WR sind in unmittelbarer Nähe der DTU, und es betraf ja beide gestern, sehr unwahrscheinlich glaube ich. Ich werde es weiter beobachten,  es war wie gesagt das erste und einzige Mal bisher.

Maschler, 2024-05-06:
@tbnobody Es ist eben wieder aufgetreten, dabei war der "Erreichbarkeit Schwellenwert" auf über 6 min hochgesetzt. Es fehlen wieder ca. 2 min.
[image]

In den Ereignissen ist leider nichts zu sehen.
[image]

Maschler, 2024-05-07:
@tbnobody Seit dem ich es beobachte tritt es täglich auf. Die Zeit ist dabei auf 12-14:30 einzugrenzen. 

Meinst du, dass ich ein Upgrade zur neuer Firmware durchführen sollte? Vielleicht hilft es oder das Problem bleibt bestehen.

mroenne2022, 2024-05-07:
@Maschler : Bei mir bisher nicht mehr aufgetreten, nur dieses eine einzige mal. Ich hab gestern den Update auf die neue Version erledigt.

relama, 2024-06-05:
Bei mir ist der Fehler (ebenfalls Web-GUI) Tagesertrag 0 bereits mehrfach aufgetreten - eher abends und kurz vor dem PV Produktionsende. 
Verwende seit ca einem Jahr den selben HMS-1000 und derzeit DTU v24.5.6 (könnte aber mit eine etwas füheren Version ebenfalls aufgetreten sein). 

LG
Reinhard

tbnobody, 2024-06-05:
> eher abends und kurz vor dem PV Produktionsende.

Das ist ein bekanntes Thema. Bitte FAQ zum Thema "Why get the daily yield values sometimes reset to zero at the evening" lesen.

relama, 2024-06-05:
Danke für die rasche rückmeldung -> ich habe jetzt die "Tagesertragskorrektur" aktiviert.

closed 2025-04-18 by github-actions[bot]