tbnobody/OpenDTU #1960: Prometheus API return wrong information in function addPanelInfo

Paddy65305, 2024-05-03:
### What happened?

I have two Hoymiles and it seams that in the promotheus API the information how are writen from the "WebApiPrometheusClass::addPanelInfo" are wrong.
I get value for opendtu_PanelInfo, opendtu_MaxPower from the wrong inverter.
The API return for the PanelInfo from INV1 the PanelName from INV2 and
The API return for the PanelInfo from INV2 the PanelName from INV1.
Same for the MaxPower.

### To Reproduce Bug

Set in OPENDTU- Config  the panelnames for two DTU and call
/api/prometheus/metric

### Expected Behavior

The API should return for the PanelInfo from INV1 the PanelName from INV1 and
The API should return for the PanelInfo from INV2 the PanelName from INV2.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.4.12

tbnobody, 2024-05-04:
Don't know. Here it works exactly as expected. Could you please provide screenshots of your config and of the output of the metrics endpoint?

Paddy65305, 2024-05-04:
I hope it's okay that the OpenDTU Screenshoot of the configuration of the Inverters are in German.
But you see in the config of my HM-300 is one string and the Hoymiles HM 600 have two.

[image]
[image]

and this will return from the api-endpoint:
[image]

the second stringname from the HM-600 is missing

tbnobody, 2024-05-04:
> I hope it's okay that the OpenDTU Screenshoot of the configuration of the Inverters are in German.

Of course :)

Thank you!
Found the issue. Will befixed in next release!

commit 7d2fb349 (2024-05-06): Fix #1960: Prometheus API return wrong information in function addPanelInfo
closed 2024-05-06 by tbnobody